rust-embedded/wg #150: Embedded WG meeting, August 13

japaric, 2018-08-06:
[6-7 PM CEST][0] on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2018-8-13,240,b8jj

Backlog.

- "Could we briefly mention talking to chip companies about Rust?" -- @thejpster
- "Managing operator status on #rust-embedded (currently it's just @japaric)" -- @thejpster?

Agenda.

- New meeting time to accommodate new members.

- Community plan https://github.com/rust-lang-nursery/embedded-wg/issues/148

- [RC1 milestone](https://github.com/rust-lang-nursery/embedded-wg/milestone/3)

- [Issues nominated for discussion, if any](https://github.com/rust-lang-nursery/embedded-wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

gbip, 2018-08-11:
Can I suggest to create a mailing list or something for embedded users to ask questions ?

I needed some help on a project and I had to make a reddit post because I had not idea where to ask it.
Reddit is not a bad thing, but I don't think there is many embedded people in there.
Thanks !

gbip, 2018-08-11:
I would prefer something asynchronous  such as mail, or github issues :smile:

japaric, 2018-08-11:
The following issues have been nominated for discussion. If you can make it to the meeting please leave your comments in the corresponding issue.

- New meeting time. #161 
- Curated list of projects. #147 
- Communicating with the teams #153 
- Community plan #148 

---

@gbip would a forum (like https://users.rust-lang.org/ but for embedded stuff) work? It has been proposed in #148.

gbip, 2018-08-11:
@japaric I think that this is a great idea, having a place dedicated to this kind of interactions !

japaric, 2018-08-17:
This meeting happened! The minutes and the IRC logs for this meeting were added in #177. 

The agenda for the next meeting can be found in #178

closed 2018-08-17 by japaric