rust-embedded/wg #152: Transfer riscv-rt repo to rust-embedded org

dvc94ch, 2018-08-07:


dvc94ch, 2018-08-12:
Marking only use stable features as wont-fix because there's not much point if you have to build a gcc riscv toolchain from source...

Moved `add a cargo template for riscv to the riscv-quickstart` repos https://github.com/riscv-rust/riscv-rust-quickstart/issues/3

closed 2018-08-12 by dvc94ch

japaric, 2018-08-12:
@dvc94ch I have enabled branch protection in the riscv and riscv-rt repos. Now everything must go through PRs and bors before landing.

> Marking only use stable features as wont-fix because there's not much point if you have to build a gcc riscv toolchain from source...

You mean there won't be an option to use these crates on stable? I would ask you to reconsider: there are some people that value stability very highly and would be OK with depending on a C toolchain if that means their program won't break overnight -- it would be great if these crates had an option to work on stable.

dvc94ch, 2018-08-12:
@japaric is there a way to leverage the built-in llvm as an assembler? Also I don't mind reviewing PRs if someone else wants to do it...

japaric, 2018-08-13:
@dvc94ch that would be `asm!` and `global_asm!` but they are unstable because makes no stability promise about their assembly syntax.

dvc94ch, 2018-08-13:
I can understand why asm! isn't stable since it depends on the llvm-ir. But global_asm! uses plain assembly which should conform to the target architecture's assembler manual. Other options would be distribute the llvm-mc with rust or porting llvm-mc to rust https://github.com/llvm-mirror/llvm/blob/master/tools/llvm-mc/llvm-mc.cpp

danc86, 2018-08-19:
Anyway from what I can tell there is no such standard C API for manipulating the special registers in RISCV. The answer is "just use inline asm". :grimacing: 

In light of that, would it make sense to pursue the idea of LLVM intrinsics for RISCV? I'm still unsure how they could be exposed in Rust though.

dvc94ch, 2018-08-19:
I think @japaric meant writing the equivalent of the CMSIS API in rust and add that to for example libcore so that it doesn't matter if there is any inline assembly.

> In light of that, would it make sense to pursue the idea of LLVM intrinsics for RISCV? I'm still unsure how they could be exposed in Rust though.

I think we should wait for the cortex-m crate to do this. They do a lot of work for us by opening RFC's etc. xD

danc86, 2018-08-19:
Ahh gotcha. It's okay to use inline asm in the *implementation* of core, even in stable Rust, because it's not exposed directly to users.

Why does the Rust implementation need to follow a "standard" C API from some vendor toolchain? In the case of RISCV where no standard C API exists, could we just make one up?

Anyway yes I agree, it seems like we should let the cortex-m folks blaze the trail for us here. :-)

japaric, 2018-08-19:
> These are just some functions/macros that are built into the compiler, or provided by some special headers included with the toolchain?

In the case CMSIS they are headers that provide the API implemented on top of whatever the underlying compiler supports (e.g. the assembly macro). See https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/Core/Include/cmsis_gcc.h , which is a reference implementation.

From what I understand ACLE is supposed to be supported at the compiler level but I never have used a compiler that supports ACLE so I'm not sure.

> Why does the Rust implementation need to follow a "standard" C API from some vendor toolchain?

Because there's a precedent: there's an [accepted RFC](https://github.com/rust-lang/rfcs/blob/master/text/2325-stable-simd.md) that says that we can add SIMD API to core::arch *if* it exactly matches a vendor API. Thus we only need a simpler RFC to add non-SIMD Cortex-M API that matches a vendor specification (e.g. CMSIS or ACLE) to core::arch.

> could we just make one up?

You can but it would be harder to get an RFC for that accepted. i.e. the Rust team may prefer not to accept a custom API and instead way for a vendor spec to appear.

dvc94ch, 2018-08-19:
> Why does the Rust implementation need to follow a "standard" C API from some vendor toolchain? In the case of RISCV where no standard C API exists, could we just make one up?

> You can but it would be harder to get an RFC for that accepted. i.e. the Rust team may prefer not to accept a custom API and instead way for a vendor spec to appear.

That's exactly what I was just writing...

danc86, 2018-08-19:
Makes sense, thanks for explaining.