rust-embedded/wg #161: New meeting time

japaric, 2018-08-11:
I send out a doodle to @rust-embedded/all asking for their availability for weekly meetings. So far, the following people have responded:

- @adamgreig (Cortex-M)
- @awygle (MSP430)
- @hannobraun (HAL)
- @korken89 (Cortex-M)
- @ryankurte (Tools)
- @therealprof (Cortex-M, HAL, Resources, Tools)
- @japaric (Lead)
- @cr1901 (MSP430)

And we have found out these three time slots where all of them can attend:

- [Monday 8 PM CEST](https://everytimezone.com/#2018-8-13,360,b8jj)
- [Tuesday 9 PM CEST](https://everytimezone.com/#2018-8-14,420,b8jj)
- [Wednesday 8 PM CEST](https://everytimezone.com/#2018-8-15,360,b8jj)

Can the rest of @rust-embedded/all make it to any of these time slots? We are missing an answer from 
@rust-embedded/riscv and from the IRR team (cc @jamesmunns).

There's still time to answer. The new meeting time won't be made effective until the week of August 20.

cc @jcsoo @thejpster @dylanmckay @Emilgardis

jamesmunns, 2018-08-11:
Added my availability to the doodle. @japaric you mentioned the IRR team, would you like me to share the link to the doodle with Hideki and Vaishali?

japaric, 2018-08-12:
@jamesmunns yes, please. I think it would be good if they attended the meetings.

danc86, 2018-08-12:
Sorry I am way too far out of everyone's timezones to make these meetings. :-(

For RISCV you really want @dvc94ch anyway, he is the expert :-)

ithinuel, 2018-08-12:
FWIW these slots are better for me as they are out of office ours. (UK).

thejpster, 2018-08-12:
When you say CEST, do you mean Europe/Berlin (i.e. you'll track local time back to CET whenever Berlin does that)? If so, I can make Monday but not the other two and I might be late because that's the tail end of my commute (but it's a bit too late to stay in the office).

Emilgardis, 2018-08-13:
These times work for me. Mondays could be tricky however for me to give full attention to.

japaric, 2018-08-13:
We now have answers from 14 people. There are 4 possible schedules:

- [Monday 8 PM CEST](https://everytimezone.com/#2018-8-13,360,b8jj)
- [Tuesday 8 PM CEST](https://everytimezone.com/#2018-8-14,360,b8jj)
- [Wednesday 8 PM CEST](https://everytimezone.com/#2018-8-15,360,b8jj)
- [Thursday 8 PM CEST](https://everytimezone.com/#2018-8-16,360,b8jj)

But at most 12 people can make it to any of these meetings. The issue [description](https://github.com/rust-embedded/wg/issues/161#issue-349751974) has been updated to indicate who can attend which schedule.

@dvc94ch @posborne @nastevens can you make it to any of these schedules? We are missing representation from the RISCV and Embedded Linux teams.

posborne, 2018-08-14:
I think the only day that would currently not work for me (recurring meeting at that time already) is Tuesday (but I could probably make it work as well).

ryankurte, 2018-08-18:
@japaric thanks for making the calendar! Much easier to keep track of imo. Are there meant to be two meetings / could it be clearer what each is if so?

[image]

japaric, 2018-08-18:
@ryankurte no, the one that shows on Tuesday for you is the old meeting time; it seems I didn't remove it properly. It should be gone now.