rust-embedded/wg #163: Relocate Newsletters to their own Repo?

jamesmunns, 2018-08-12:
Do we want to move the newsletters to their own repo? This would take effect after today's (2018-08-12) newsletter.

Downsides:

* There isn't much throughput to that repo, right now we only have a newsletter every two weeks, though this may increase to once per week.
* Another repo to follow in the organization

Upsides:

* Changes to the newsletter doesn't clutter up the main `wg` repo
* It would be easier to set up a static site (`https://newsletter.rust-embedded.org`), if we want at some point.

CC @japaric @rust-embedded/resources

japaric, 2018-08-12:
If we agree on making a website / blog as discussed in #148 the newsletters should be moved there and rendered as old blog posts, IMO.

japaric, 2018-08-28:
Done in #198

closed 2018-08-28 by japaric