rust-embedded/wg #169: Pre-RFC: "Partnership" Team to address business outreach topics

jamesmunns, 2018-08-13:
We would like to consider forming a team to be a public point of contact for potential business contacts.

This could include any of the following:

* Be the first point of contact for business parters
* Developing an official HAL for Silicon vendors
* Officially promoting embedded rust, from a business perspective

We will need to solicit feedback on the following topics:

* What should be the name of this group?
* What should be the core responsibilities of this group?
* Who should be a member this team?

jamesmunns, 2018-08-13:
> NOTE: we have not decided what level of disclosure is necessary, so do not take this comment as a normative example. I just wanted to share what I am doing, as it came up in an IRC meeting

Personal Disclosure: I am in the process of starting a consultancy company, focused specifically on Rust for developing systems/embeddded level software, and training on the same topics. I'll update this comment with a bit more detail after we announce this publicly at RustConf (not a big secret, the company just hasn't launched yet :smile:)

Regarding Issue #170, I intend to offer "development of HAL/BSP"s as a possible service I would like to offer, either to silicon vendors, to PCB/prototyping board manufacturers, or individual companies that need or want support for their platforms.

thejpster, 2018-08-13:
I'd be happy to join this.

jamesmunns, 2018-08-13:
I would be happy to join and/or lead this, with the disclosure above in mind.

thejpster, 2018-08-13:
Possible names:
* rust-embedded/promotion (promotion@rust-embedded.org)
* rust-embedded/marketing (marketing@rust-embedded.org)
* rust-embedded/outreach (outreach@rust-embedded.org)
* rust-embedded/enterprise (enterprise@rust-embedded.org)
* rust-embedded/pr

thejpster, 2018-08-13:
Based on how I would re-act to those addresses landing in my Inbox, I'm actually leaning towards public-relations@rust-embedded.org.

thejpster, 2018-08-13:
I'd also add to the list at the top of the issue:
* Organising / co-ordinating EWG representation at public events (e.g. trade shows, conferences, digital making festivals, etc.)

chrysn, 2018-08-13:
Where discussion is public, this team should keep a list of pending requests to the vendor for Rust support or Rust related SVD file improvement. That list should contain both link to requests from others and, when the team starts actively pushing, own requests. It would need to have a low threshold for third party contribution ("edit on github" link to create a PR to add a line or mark a discussion as obsolete/concluded).

japaric, 2018-08-13:
cc @rust-embedded/all

hannobraun, 2018-08-14:
I agree that establishing such a team is a good idea, and I'm interested in participating. Since I'm running my own Rust-focused business ([disclosure here](https://github.com/rust-embedded/wg/issues/170#issuecomment-412857930)), this overlaps with what I'm already doing marketing-wise.

We should probably finish discussing #170 before making any decision here. I think it's obvious that this team would be talking to people that I and others on the team would want to be going into business with. I don't see this as a problem, as long as
- nobody is misrepresenting themselves as speaking for the WG when they aren't and
- nobody is influencing WG decisions without disclosing their outside interests.

@jamesmunns 
> Developing an official HAL for Silicon vendors

Could you clarify what you mean by that? I don't fully understand how doing development would fit into the work of the team.

@thejpster 
> Possible names:
> [...]

I like "Outreach", although I think "Public Relations" is also okay.

@thejpster 
> I'd also add to the list at the top of the issue:
> * Organising / co-ordinating EWG representation at public events (e.g. trade shows, conferences, digital making festivals, etc.)

Agreed.

By the way, I'll be at [FrOSCon](https://www.froscon.de/) next week. There'll be a rust booth (not organized by me), and I'll be present, showing off Rust running on various microcontrollers.

dantho, 2018-08-23:
Core Responsibility:  Identify and/or solicit Rust advocates inside partner companies.
Internal advocates are in a better position to "Promote" Rust within a company.  (Full Disclosure:  I aspire to be one of my company's Rust advocates.  Actually, I'm already an advocate, but I'd like it to be official.  I want a badge.  Or a sticker.)

Core Responsibility:  Develop and maintain a "Why Rust" ~whitepaper~ pitch deck for corporate advocates to reuse in their internal presentations.  Solicit feedback on same and update frequently.

Core Responsibility:  Track Rust usage by companies, whether or not they are "partners".  Somehow validate usage, don't just pass along rumors.  (Do track rumors, then try to confirm them "on the record".  Trade shows are good for this.  How about a Rumor Bounty for the team member who validates first?)  Push for disclosure on Rust success stories -- public is best, but behind the corporate firewall is a great start (and usually precedes public disclosure.)  The public stories become data in the "Why Rust" pitch deck.

Core Responsibility:  Maintain a list of open source or other training materials (and vendors) for use by partner companies wishing to upgrade Rust literacy.

Advice for team members:  Don't treat "a" company as a single entity.  Consider each company as a collection of entities -- they are.  Each group within "a" company might not know what the others are doing.  (Also, they might not care that much.)  In other words, identify multiple advocates and track multiple success stories -- you might need all of them.

therealprof, 2024-06-11:
@jamesmunns Is this something you'd still like to pursue?

jamesmunns, 2024-06-11:
@therealprof this is definitely still relevant, and a thing I'd like to discuss as part of the discussions with the Rust Council, and having some subset of the EWG be part of the Rust project.

I don't think it'll be in the same shape as I originally proposed, but happy to keep this issue and revisit it soon-ish.