rust-embedded/wg #170: Pre-RFC: Extend CoC to cover Conflict of Interest Disclosure and Rules

jamesmunns, 2018-08-13:
As the market for embedded Rust expands, more developers involved with the Working Group will have a business interest in supporting and expanding the embedded Rust efforts.

At the current point, some members of the Embedded WG have personal or full time businesses focused on the embedded rust domain.

This RFC would aim to address the following topics:

* What possible conflicts of interest must be disclosed, and to what extent, as being part of the embedded working group?
* What rules or guidelines should be put in place, to avoid possible perception of conflict of interest?
* How should these rules, guidelines, and/or disclosures be captured?

thejpster, 2018-08-13:
To give an example, I think it would be wrong if someone was seen to be saying something to a third-party on behalf on the EWG which was clearly in that person's interest and not clearly in the EWG's interest. We're here to make Rust better, not to serve ourselves, although where the EWG's interest and our personal interests align, then I think that's fine. It would be odd if people on the EWG were barred from profiting from something they clearly take a lot of interest in and have a lot of knowledge about.

thejpster, 2018-08-13:
Perhaps a rule such as "No member of the EWG shall claim to represent the EWG or to be acting on behalf of the EWG without an approved RFC giving them the authority to do so. Stating your own membership of the EWG and/or the ability to represent an third-party interest at EWG meetings or in EWG documents and issues is expressly allowed, provided that those third-party interests (if any) are made clear to the EWG in the Register of Interests."

jamesmunns, 2018-08-13:
NOTE: we have not decided what level of disclosure is necessary, so do not take this comment as a normative example. I just wanted to share what I am doing, as it came up in an IRC meeting

Personal Disclosure: I am in the process of starting a consultancy company, focused specifically on Rust for developing systems/embeddded level software, and training on the same topics. I'll update this comment with a bit more detail after we announce this publicly at RustConf (not a big secret, the company just hasn't launched yet smile)

Regarding Issue #170, I intend to offer "development of HAL/BSP"s as a possible service I would like to offer, either to silicon vendors, to PCB/prototyping board manufacturers, or individual companies that need or want support for their platforms.

japaric, 2018-08-13:
cc @rust-embedded/all

hannobraun, 2018-08-14:
I'm self-employed, and I've been working on focusing my business solely on embedded Rust for a while now. I offer ARM Cortex-M firmware development in Rust, which includes development of HALs, BSPs, drivers etc. (more info at https://braun-robotics.com/)

Not sure if that counts as disclosure, as that's been going on longer than my involvement in the working group. Although it is only very recently that I've actually started making money with this.

I think clarifying the rules around conflicts of interest and representing the working group is a good idea.

therealprof, 2024-06-11:
As briefly discussed in our triage meeting, I think time has shown that there's pretty much no overlap between WG interests and commercial interests which would invoce a conflict of interest.

@jamesmunns expressed that it might be a good idea to note that the majority of the leaders in the core team should not be allowed to work for the same employer.

I don't think that this would cause any problems but I'm keeping this one open to allow us to discuss and address this point if required.