rust-embedded/wg #171: IRC logger hosted or maintained by us

japaric, 2018-08-13:
We have been used the logging service provided by the Tock project: [tockbot](https://bot.tockos.org/tockbot-moz/embedded-bot/), but it was down for the past week and it seems to be down again. We discussed in the last WG meeting switching to an IRC logger hosted or maintained by us and that has more people that can look after it.

Options:

- We could request [botbot.me](https://botbot.me/) to set up and maintain the logging service. I don't know what's their average uptime.

- We could host a [botbot instance](https://github.com/BotBotMe) ourselves. Tockbot is a botbot instance.

- We could use something else. @thejpster mentioned that the selenium projects [logs everything to GitHub](https://github.com/SeleniumHQ/irc-logs/tree/master/2015/09).

Any solution you would recommend? Or should we stick to one of the above solutions?

cc @rust-embedded/all

levex, 2018-08-13:
I like the idea of hosting the logs on GitHub a lot! I can imagine myself cloning the repo and `grep`ing through it for some information without having to use an often clunky web UI.

BotBotMe seems to have not been active for the past few months, what about using the following bot?
https://mozilla.logbot.info/

It seems to be part of Mozilla IRC, already.

adamgreig, 2018-08-15:
:+1: for using logbot, it seems like an easy solution and already in use by a bunch of rust channels.

japaric, 2018-08-17:
Using logbot SGTM

japaric, 2018-08-20:
To move things forward here I'd like to propose (a) starting out with logbot and (b) postpone revisiting doing our own thing.

I'm going to nominate voting on the above proposal during the next meeting.

commit 3ce9b01f (2018-08-23): Merge #190

190: use logbot to log our IRC channel r=japaric a=japaric

This a proposal to log our IRC channel using logbot.

This proposal (PR) needs at least 10 votes (approvals) from @rust-embedded/all
to be accepted. After the proposal has been accepted we'll proceed to implement
it.

@rust-embedded/all please vote on this proposal using [pull request reviews].
Or if you have a concern leave a comment.

[pull request reviews]: https://help.github.com/articles/about-pull-request-reviews/

closes #171
PR #190 (2018-08-23): use logbot to log our IRC channel
This a proposal to log our IRC channel using logbot.

This proposal (PR) needs at least 10 votes (approvals) from @rust-embedded/all
to be accepted. After the proposal has been accepted we'll proceed to implement
it.

@rust-embedded/all please vote on this proposal using [pull request reviews].
Or if you have a concern leave a comment.

[pull request reviews]: https://help.github.com/articles/about-pull-request-reviews/

closes #171
closed 2018-08-23 by bors[bot]