rust-embedded/wg #176: Logo color

andre-richter, 2018-08-17:
I recently wondered if the blue color that we use for the embedded WG logo is an "official color".

It seems that the standard Rust logos all use black. Is that something we want to align to?

hannobraun, 2018-08-17:
I don't think there's anything wrong with standing out a bit, and I really like the current logo. On the other hand, I know nothing about logos and design, so maybe just ignore me :grin:

ithinuel, 2018-08-17:
Both black & coloured rust logo are on https://www.rust-lang.org/en-US/legal.html so I think both are official logos :)

osterwood, 2018-11-27:
Generally trademarks are done for the black & white logos only, as they cover all colorized variants.  A color trademark only protects that particular coloring of the logo.  If the color is an essential element of the logo, it makes sense to file a second application for that particular color treatment -- otherwise not worth it.

stappersg, 2020-02-27:
@andre-richter is your question answered?  [Yes/No]

@andre-richter are you aware that you have the power to close issues that you have created? [Yes/No]

At all:  For which reason is this issue still open?

closed 2020-02-28 by andre-richter