rust-embedded/wg #179: Request for Feedback: Presenting Embedded Rust at a Conference Booth

hannobraun, 2018-08-17:
Hi everyone, I could use your help.

I'll participate in the Rust booth at [FrOSCon](https://www.froscon.de/en/) in Bonn, Germany next week (25th and 26th). There are other people involved, but I'll be the one representing the embedded Rust community there.

I'm currently working on some written material for the booth and would love to get some feedback. I'm sure I'm missing something obvious.

First, my general plan:
- I'm going to display 3 boards running Rust (STM32F0 Discovery, LPCXpresso824-MAX, a breadboard with an LPC824).
- Unfortunately all just run variations of "blinking an LED". I don't have time to prepare something more impressive and don't have anything better at hand.
- For each board, I'll have a piece of paper with basic information (name of the board, size of flash/RAM, a few words about the program it's running).
- There'll be an additional piece of paper with some general information about embedded Rust. This is the main thing I'd like feedback on.

Before I show you what I've come up with so far, 2 notes:
- I don't want to have a lot of words. Just some basic information in big letters.
- The final information will have to be in German anyway, so I didn't put any effort into the writing that's about the follow.

Here's what I had in mind for that main piece of paper:
- Header: Embedded Rust (insert logo here)
- Rust allows you to write safe and reliable code even on tiny systems with just a few kB RAM
- Doesn't need a heap or a runtime; no overhead compared to C
- Growing ecosystem: Check out Awesome Embedded Rust (https://github.com/rust-embedded/awesome-embedded-rust)

Some questions to the public:
1. Is there anything I'm missing? Keep in mind, I don't want to write a novel, and I will be there to provide more information to those interested.
2. Any suggestions for an URL where I can point people? I chose Awesome Embedded Rust, but I feel there are no really good options right now.
3. Any other suggestions?

Thank you to everyone who's going to take the time to help me!

ithinuel, 2018-08-17:
From my experience at the Raspberry Field event in Cambridge, the [Monotron](https://github.com/thejpster/monotron) by @thejpster is very efficient to attract people and make them ask question :)

jamesmunns, 2018-08-18:
@skade has talked about getting a monotron together for froscon

hannobraun, 2018-08-20:
@ithinuel @jamesmunns 
I don't have a Monotron, don't have time to build one, and even if I had one, I don't think I could get a VGA monitor there. I hope we'll have one there, but I fear I won't be much help.

@killercup 
Thanks for mentioning this here, I fully agree. I'll try to create a design today, but I'm not sure if I'll get to it. If anyone wants to take a stab at this, I'd appreciate it!

skade, 2018-08-20:
I would have some time to build one, but I don't have equipment or a VGA screen in Hamburg :(.

killercup, 2018-08-20:
@hannobraun no need to create a design per so! If you have a bunch of links (an amount similar to what the other cards have) and you can tell me how to arrange them, I can do the rest :)

hannobraun, 2018-08-20:
@killercup Thanks. I'll look into it as soon as I can, do as much as I can manage, and leave the rest to you :)

japaric, 2018-08-21:
@hannobraun the embedded book is in a pretty rough state at the moment. I would either omit it, or note that it's in an early stage and link to the repository instead.

killercup, 2018-08-21:
@hannobraun thanks! That's all I really need! Here'a quick first draft:

[image]

hannobraun, 2018-08-21:
@killercup Thank you, this looks great!

A few notes:
- As per @japaric's suggestion, we should replace `book.rust-embedded.org` with `github.com/rust-embedded/book`.
- You omitted the link to the working group under `Community`. By accident or on purpose?
- The link on the bottom says `rustwasm.github.io`. Should probably say `rust-lang.org` (http://www.rust-embedded.org/ is not ready).

killercup, 2018-08-21:
@hannobraun updated! (totally forget that link after copy-pasting the WASM stuff)

[image]

[image]

hannobraun, 2018-08-21:
@killercup 
Thanks, I like it!

Just noticed the edit under your post above:
> Also, would it make sense to link to https://crates.io/categories/embedded instead of crates.io itself?

I don't think so. The embedded category is probably too narrow, as much from https://crates.io/categories/no-std also applies, for example. I don't feel strongly about this, but I would just leave the link to crates.io as it is.

@japaric 
The latest draft links to the repository of the embedded book, but leaves out the note that it's a work in progress. I think that's appropriate, because the repository page mentions that the book is a work in progress, and because the state of the book might change while we still have copies of these cards to give out. Do you agree, or do you want the `(work in progress)` to go back? Both are fine with me.

killercup, 2018-08-21:
FYI I omitted the "WIP" suffix because I changed the headline above the book link to "Contribute to the community book!"

hannobraun, 2018-08-21:
Oh, totally missed that. Fine with me!

killercup, 2018-08-21:
FYI opened https://github.com/rust-community/resources/pull/13 with the pdf version of what I posted above. I'll try to get this printing *tomorrow*. Let me know if want any more changes!

skade, 2018-08-22:
I have tasked someone in my company with building a Monotron over the day. I'd be happy if someone could source a VGA monitor, preferably 4:3.

hannobraun, 2018-08-22:
I could get a VGA monitor, but I really don't want to: First, it's currently located in a village with spotty bus connection, and I don't have a car. Second, I'd still have to carry it all the way to the conference and back (~3 hours each by train, multiple connections). It would be great if someone else could bring one.

killercup, 2018-08-22:
I talked with Florian, I'll bring a 2000-era TFT screen. Not the same kind of retro, but 4:3 at least ;)

ithinuel, 2018-08-22:
That's what we had at Raspberry Field, It works very well ! 👍 

[image] (from twitter).

hannobraun, 2018-08-28:
FrOSCon is over, and it was a huge success! Lots of interest at the booth.

I've published the code for the embedded demos that ran on my boards: https://github.com/hannobraun/froscon2018-demos

Thanks to everyone who helped!

closed 2018-08-28 by hannobraun