rust-embedded/wg #193: How to deal with voting overhead?

japaric, 2018-08-22:
In the last meeting it was mentioned that as the /all team grows it will take longer to approve proposals that require a voting majority from /all. /all currently has 19 members (and 2 pending invitations) so proposals need at least 10 votes to be approved.

This issue is to discuss ways to prevent this from becoming a problem in the future. We'll discuss this synchronously in the next meeting.

Any ideas?

cc @rust-embedded/all

adamgreig, 2018-08-25:
At the moment we don't seem to have the person who opened the PR vote on it, but assuming they are also a member of the appropriate group/the WG, should they be allowed to approve it as well? This might help push a number of RFCs over the majority threshold, and as far as I can tell is still in line with our majority definition.

japaric, 2018-08-28:
Something I didn't get to mention during the meeting is that I think that notifying the people that they need to vote is a problem right now because the notifications get lost among other GH notifications.

So how about sending e-mail titled e.g. "[Action required] japaric has requested you to vote on this rust-embedded/wg issue" when we start a new voting process? Ideally this should be automatic but at least having a template to do it manually for now would be a good start. This would need the "@teams.rust-embedded.org" e-mails so it would be blocked on implementing that. And if we do the exponential backoff thing we could send new e-mails on the subsequent weeks to the people that have not yet voted.

japaric, 2018-11-20:
RFC #206, which addresses this issue, has been approved and merged.

closed 2018-11-20 by japaric