rust-embedded/wg #217: Embedded WG meeting, October 2

japaric, 2018-09-26:
[8-9 PM CEST][0] on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2018-10-2,360,b8jj

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

Agenda.

- Triaging [edition issues](https://github.com/search?q=org%3Arust-embedded++is%3Aopen+milestone%3A2018&type=Issues)

- [Issues nominated for discussion, if any](https://github.com/rust-lang-nursery/embedded-wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

japaric, 2018-09-29:
@ryankurte I personally think that such change would be fine to do as a PR that the whole ops team has to approve to land. Also give it like a week before merging to let other raise concerns, if any.

gbip, 2018-10-06:
Is there any logs for this meeting ? :smile:

japaric, 2018-10-09:
This meeting happened! The minutes and logs can be found in PR #225.

The agenda for next meeting is in issue #226.

closed 2018-10-09 by japaric