rust-embedded/wg #221: Make the Rust libstd an explicit dependency.

mwarning, 2018-09-28:
Making a the Rust libstd an explicit dependency might have advantages for Rust:

 * other std libs would become more equal first class citizens
 * encourage crate developers to not let a library depend on the rust lib std by default
 * no `#![no_std]` needed anymore, e.g. include stdc as dependency
 * easier search for crates that do not depend on the Rust libstd

mwarning, 2018-09-28:
I hope this does not come over as too uninformed. I am still a Rust noob.

jamesmunns, 2018-09-30:
Hey @mwarning, this suggestion would go a bit above what the embedded wg would be able to change.

I do think you make good points, however this would also come with the downside of breaking a large amount of existing code, which would mean it would take a while before it would happen (they would have to announce deprecation in one edition, then make that in the next edition).

If you are interested in taking this forward, I would suggest that you check out https://github.com/rust-lang/rfcs, and follow the RFC process (as this would affect the entire language, not just the embedded section).

I'm going to go ahead and close this for now, feel free to ask if you have any further questions.

closed 2018-09-30 by jamesmunns

mwarning, 2018-09-30:
ok, thanks!