rust-embedded/wg #224: Transfer ownership of thestm32f103xx-hal repository

gbip, 2018-10-07:
Currently there is a lot of PR (more than 13) that are awaiting a merge or a review.
It would be great if you could transfer the ownership (or at least give merge rights) of the repository to someone who has the time to maintain this crate.
Thanks a lot !

cc @TeXitoi

japaric, 2018-10-07:
> It would be great if you could transfer the ownership (or at least give merge rights) of the repository to someone who has the time to maintain this crate.

Are you volunteering? :-)

I don't have to write an structured answer so here's a bullet list.

- I mentioned in the last meeting that I'm happy to hand over the repo. The receiving entity doesn't necessarily have to be the WG. cf. stm32-rs and nrf-rs orgs.

- Some people have concerns about having any HAL implementation under the WG but I'll let them voice their concerns themselves.

- If we do decide to take a HAL impl crate into this org, IMO, it makes more sense to take the stm32f30x-hal since it's used in the embedded book and the discovery book. The stm32f103xx-hal has always been more of experimental grounds (see its list of git dependencies) whereas the stm32f30x-hal has been more of the reference implementation (stable bits).

cc @rust-embedded/hal @rust-embedded/cortex-m

therealprof, 2018-10-07:
@gbip 

> What would be the benefits of taking this HAL impl into this org ?

A group of people with the necessary interest and experience handling this.

@japaric 

The stm32f103 is the most used and easiest accessible MCU from STM which is the reason why it receives such a lot of attention. I don't see the current way of having one HAL crate per MCU family member as the way to go since the peripherals work exactly the same within a family and only the number and options differ between individual family members (and package) types. I'd rather see this turn into HAL crate which works for the whole family at once, similarly to what I started with the proof-of-concept https://github.com/therealprof/stm32f4xx-hal crate, made possible by @adamgreig's fantastic stm32-rs crate. I'd do the same with this crate (and drop the peripheral registers crate, too), in fact I already attempted that a few months back but then decided to go for the higher families due to requiring Ethernet support, which I cannot even get to work with the official STM HAL on the STM32F107...

TeXitoi, 2018-10-07:
The logs: https://mozilla.logbot.info/rust-embedded/20181002#c15405289

TeXitoi, 2018-10-07:
I'd be happy to help on something like stm32f1xx-hal, it seems the best approach.

I agree that almost everyone that I talked that is doing embedded rust have a stm32f103 somewhere, and that having a featureful hal crate that's on crates.io and working on stable is a must.

RandomInsano, 2018-10-07:
Speaking of that, I bought the discovery board specifically because of Rust Embedded but will likely never use it in a project.

In contrast, I own eight STM32F103 boards that have replaced the ATMega Arduino boards for all of my embedded projects.

gbip, 2018-10-08:
cc @TheZoq2

TheZoq2, 2018-10-08:
Having a stm32f1xx crate sounds like a great idea but I would assume it takes some work to get to that point. For those of us wanting to use our 103s now, it would be great to see more activity in that repo even if it might eventually be replaced by a more generic crate :)

As for volunteering, I'd be happy to help maintain it, but like @gbip I'm not sure if I would be comfortable doing it on my own. Perhaps a few of us could do it together? As mentioned in the other repo, @TeXitoi would also make a great candidate imo.

therealprof, 2018-10-08:
@TheZoq2 It's fairly easy to make it generic. Two steps are required: Switching the peripheral register crate to stm32-rs (which might or might not require some tweaking) and some feature gate magic.

RandomInsano, 2018-10-09:
I’m fine to handle grunt work if tasks are laid out. I’d have to get up to speed on bringing up a new board.

I have i2cdev to (hopefully) finish up but should be free by the end of October.

japaric, 2018-10-09:
To get the ball moving I have added @TeXitoi as a collaborator (I tried to transfer the repo but it was not possible due the existence of forks). If someone else wants to become a collaborator ping me in this thread.

TeXitoi, 2018-10-09:
@japaric thanks

therealprof, 2018-10-09:
@japaric Yes, please.

therealprof, 2018-10-09:
@japaric Thanks!

therealprof, 2018-10-09:
@TeXitoi Works for me. Please ping me if there's anything specific that needs reviewing ASAP.

thejpster, 2018-10-09:
As I noted on IRC, I'm generally against bringing chip specific crates in the wg. I think it sets a bad precedent, and I don't think the WG should be marking specific chips from specific vendors out for special treatment. But, I hold strong opinions lightly so I'm open to persuasion!

TeXitoi, 2018-10-09:
Having an official reference implementation of embedded_hal on a micro-controller give a good example of how such a hal should be implemented.

But stm32f103xx is not in such a state for now.

thejpster, 2018-10-10:
> Having an official reference implementation of embedded_hal on a micro-controller give a good example of how such a hal should be implemented.

If we do want a reference implementation, I suggest we write one for the target that QEMU implements because that is free to all and doesn't require anyone to buy hardware (irrespective of how 'cheap' hardware might be in some regions, it may not be so cheap in others), and it avoids singling out a particular vendor. It also makes it easier to test.

therealprof, 2018-10-10:
@thejpster I can totally not understand your negativity about sharing ownership about something people  are passionate and volunteering to care about. Isn't that what we do?

> If we do want a reference implementation, I suggest we write one for the target that QEMU implements because that is free to all and doesn't require anyone to buy hardware

How is that helping (apart from being a nice academic exercise)? We want people to use Rust on real embedded hardware and not just diddle around  in some sandbox and then give up due to lack of real world interaction. I don't think we should remove any reference to real hardware from a `particular vendor`from all our documentation, especially not the `discovery` book. 😏

Well, I guess we could still chose the same approach we did for `Nordic` chips and create a `STM` group to pool effort...

gbip, 2018-10-10:
I think that creating an STM group would be a really nice idea !

TeXitoi, 2018-10-12:
https://github.com/japaric/stm32f103xx-hal/issues/106 might interest the watchers of this issue.

TeXitoi, 2018-10-12:
Maybe we can close this issue as stm32f103xx-hal will not be moved to the wg in the near future?

therealprof, 2018-10-12:
Sounds good to me.

closed 2018-10-12 by therealprof