rust-embedded/wg #228: Write a PoC tool that generates accessors from the ARM System Register XML.

andre-richter, 2018-10-09:
As discussed in #207, we figured it is a good idea to write a tool that generates accessors from the [ARM System Regsiter XML](https://developer.arm.com/products/architecture/cpu-architecture/a-profile/exploration-tools).

We should start with a simple PoC. I'd personally find it lovely if it would leverage the `register-rs` crate, and ideally generate output that is in the current spirit of the `cortex-a` crate. This would allow us to eventually replace the register accessor parts of `cortex-a` with output from the tool.

@wizofe, you volunteered to give this a try. Still interested?

CC @rust-embedded/cortex-a for discussion.

ryankurte, 2018-10-10:
I'm not sure what the overlap is like, but, it might be interesting to look at the commonality with svd2rust and whether we can share any of the register generation between projects?

It'd be nice if the different platforms we support ended up with similar looking APIs imo (though I wouldn't suggest that should look like our current svd2rust outputs).

wizofe, 2018-10-11:
@andre-richter Yes definitely I would like to give this a go. Do you believe that `svd2rust` from @ryankurte's proposal is relevant for us? 

And if so how would you imagine such an implementation? It would be good to have a starting point on the structure of this.

parched, 2018-10-15:
I started doing a similar thing to `cortex-a` when I first started with rust, it looked like this
```
pub struct XXX_ELx(u64);

impl XXX_ELx {
    pub unsafe fn read() -> Self {
        // asm!(...)
    }
    
    pub unsafe fn write(self) {
        // asm!(...)
    }
    
    pub fn get_field_y(&self) -> u64 {
        (self.0 & Y_MASK) >> Y_SHIFT
    }
    
    pub fn set_field_y(&mut self, value: u64) {
        self.0 &= !Y_MASK;
        self.0 |= (value << Y_SHIFT) & Y_MASK;
    }
}
```

I don't think it makes sense to go any higher level that. e.g. having a `pub static XXX_ELx` only makes sense if the register isn't banked per core, this sort of thing could be done in a higher level crate like `cortex-a` with some human interpretation.

andre-richter, 2018-10-15:
I am not sure if I could grasp your statement fully.

The reason I declare the CPU registers as `pub static` in the `cortex-a` crate is that they are merely containers for inline functions. They don't encapsulate a single byte that is instantiated somewhere in memory (in contrast to the MMIO counterpart, which holds the MMIO address in the struct). So you can gain access to register functions with a small `use` at the top of your rust file, and you don't have to instantiate the register struct each time you want to use it.

andre-richter, 2018-10-15:
Tock really does an awesome job on zero-cost abstraction. I've showcased it a little here: 

https://github.com/rust-embedded/rust-raspi3-tutorial/blob/master/0C_virtual_memory/README.md

therealprof, 2024-06-11:
@andre-richter Is this still something you think we should do?