rust-embedded/wg #232: Question about ARMmbed

giannissc, 2018-10-10:
Is there any documentation to look at on how to get started on using the mbed platform. I would like to use on a university project and I thought it would be a great opportunity to practice rust!! Any help/guidance would be really appreciated!!

theotherjimmy, 2018-10-11:
Hi giannissc,

I also don't know about any document about Mbed OS + Rust. (Mbed RS? ;))

I also work on mbed, and my contributions here are not on Arm's behalf. I would also like to use Rust more often in my embedded development. I'm in UTC-5, and you can reach me through github.

giannissc, 2018-10-21:
Well I am gonna use NRF-52 development board to develop a step tracker by utilizing  an IMU (accelerometer, gyroscope, magnetometer). Do you have any advice on how to get started with this?? And thank you for your willingness to help!! :)

jamesmunns, 2018-10-21:
@giannissc come check out what we have going on at https://github.com/nrf-rs! We have crates for the nrf52832 and 52840, and have started discussing adding support for the nrf52810.

You probably want to get started with https://github.com/nrf-rs/nrf52-hal, which board are you using?

giannissc, 2018-10-21:
@jamesmunns thank you very much will check them out! ;)

giannissc, 2018-10-21:
@

jamesmunns, 2018-10-21:
@giannissc do you have the 52832 DK or the 52840 DK?

These boards aren't officially supported by the embedded WG, but a couple of us that are on the WG are working in the nrf-rs organization as we use various Nordic devices for personal projects.

I think there is an embedded Hal driver for the mpu9250, check out crates.io or the rustembedded/awesome-embedded-rust repo

adamgreen, 2018-10-24:
In what ways should Rust be integrated with mbed? I see mbed as an ecosystem like Arduino with an IDE (online and now offline with mbed Studio), build tools, C++ based HAL, C++ based Frameworks with an emphasis on IoT functionality (ie. networking, BLE, etc), community provided libraries, mbed supported boards, debugging tools, RTOS, etc.

It would definitely be cool to see embedded Rust working on all of the [mbed boards](https://os.mbed.com/platforms/). It might also be nice to get the [RTX RTOS used by mbedOS](https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/RTOS2/RTX) working with embedded Rust too. The rest of the mbed ecosystem is either very C++ specific and/or already has Rust equivalents (cargo, crates.io, Rust based HALs, etc). I do have some past experience with mbed and am interested in seeing how embedded Rust and that past experience can come together but I am still trying to get up to speed on embedded Rust. Maybe I can help you guys on whatever you decide to do with Rust & mbed together in the future.

therealprof, 2018-10-24:
@adamgreen I would assume that people are using mbed as a short term for mbedOS, meaning just the OS instead of the whole ecosystem.

giannissc, 2018-10-24:
@jamesmunns I have the 52832 board. Will check out the crates and thank you for the help! Would it be cool to ask you questions if have any trouble with it?

giannissc, 2018-10-24:
@adamgreen Yeah I am mean it doesn't have to be mbded. I care more about the hardware support and being able to compile and run rust code on them instead of C++

jamesmunns, 2018-10-24:
@giannissc I'm happy to answer questions, but my response time will likely be hit or miss :) - I would suggest opening an issue at https://github.com/nrf-rs/nrf52-hal if you have any questions, I'll see all the notifications there. Feel free to @ me on github.

hug-dev, 2019-10-18:
Sorry for bringing this topic up again but just saw this 😋 
When I last looked at it, Mbed OS provided a HAL interface that different targets have to implement in order to work with it.
Maybe it would be possible to:
* Using C to Rust FFI, make a Rust implementation of an Mbed OS HAL to have a Rust target running Mbed OS
* Make a generic "converter" between Mbed OS HAL and Rust [embedded-hal](https://github.com/rust-embedded/embedded-hal) so that every target that implement embedded-hal could easily work with Mbed OS?

theotherjimmy, 2019-10-18:
In my personal opinion, that could work. Things to be wary of are:
 * the HAL is designed to be compiled within Mbed OS.
 * parts of the HAL are designed to be used with weak symbols (this harms your ability to use a static lib).
 * Mbed OS likes to control the ISRs and delegate processing to lower exception levels.

OTOH, Mbed OS does have something called "Bare Metal" which should get you compiling something without an RTOS or any fuss. You could try adding a "Bare Metal" `mbed_app.json`, building with `--library --no-archive` to prevent linking or making a static lib. You could then scan the build directory for object files to link against with Rust controlling the main.

Let me know if you have any questions.

therealprof, 2019-10-18:
I don't see any reason why it wouldn't work. However I'm not sure why one would want to do this as you're basically getting the worst of both worlds: The unsafety of the Mbed OS abstractions, the additional complexity of having to deal not only with C code but also the Mbed OS build system, the overhead for runtime checks in Rust and all of that without the ability to optimise the generated code both on C and Rust side.

IMHO such an endeavour would only be worth it if you utilise advanced functions of an OS, like the BLE and network stacks or multitasking of Zephyr or MyNewt.

hug-dev, 2019-10-21:
Thanks @theotherjimmy for the warnings!
It was merely some brainstorming about a Rust + Mbed OS integration, and maybe an interesting topic for someone to research.

closed 2020-12-05 by giannissc