rust-embedded/wg #235: Rust presence at emBO++ conference in Bochum, Germany

hannobraun, 2018-10-12:
At the Rust booth at FrOSCon a while back, I met one of the organizers of the emBO++ conference and he recently followed up with me. He's interested in having a Rust presence at the conference, and invited me to apply for a talk or workshop.

I'm currently not interested in public speaking, but I might be interested in giving a workshop about embedded Rust. I won't do it alone, however. Would anyone be interested in doing a workshop with me, or maybe talking about Rust at the event?

Some dates and facts:
- [emBO++](http://embo.io/) is an embedded conference in Bochum (Ruhr area), Germany.
- It will take place from March 14 to 17 2019.
- [Call for papers](https://www.embo.io/fileadmin/Content/PDF/emBO_callforpapers.pdf) ends December, 31st.
- Options are a 10-, 15-, or 50-minute talk, or a 3.5 hour workshop.
- There are also booths, but I don't know about the conditions there. Can find out, if there's interest.
- More info in the [call for papers](https://www.embo.io/fileadmin/Content/PDF/emBO_callforpapers.pdf)

At this stage I'd like to gauge if there's any interest. If there is, we can decide how to move forward.

I'm going to ping a few people now, that are local to the Ruhr area, WG members based in Germany, or that might be interested for other reasons. My apologies if I'm unintentionally spamming someone.
cc @mre, @killercup, @jamesmunns, @therealprof, @skade, @badboy

This is certainly not a complete list, so feel free to pull more people in, if you know someone that might be interested.

@korken89 I heard you might be at this event?

mre, 2018-10-12:
I'd be interested in doing a workshop with you. Let's talk. 😊

hannobraun, 2018-10-12:
@mre Nice! :)

I'm won't be home most of the weekend. Let's get in touch next week.

thejpster, 2018-10-13:
Do they cover travel and accommodation? They didn't say they did on the CFP.

hannobraun, 2018-10-16:
@thejpster 
> Do they cover travel and accommodation? They didn't say they did on the CFP.

I assume not, or they probably would have mentioned it in the CFP. (I mean, they do specifically mention the ticket and dinner.)

korken89, 2018-10-28:
Hi all, sorry for the late reply.

I have been a speaker at emBO++ previously and I can strongly recommend going there, it has great exposure to the current embedded C++ community.
I will send in a workshop proposal (sadly not Rust based), but if we have ideas for good Rust talks I am open to send in talk proposals as well, I am a bit low on ideas from my recent absence.

If people want to talk or discuss what could be a good talk, I'm all in!
Also, I had a talk with one of the organizers, just a talk showcasing  Rust in an embedded project would be good enough - they are quite open to see Rust and how it stacks up to C++.
With that said, if the talk includes comparisons to C++ it would be a perfect talk.

On travel and accommodation, last year they did not due to low interest from sponsors.
This year it might change, but I have not definitive answer. I will be going in any case, from my own pocket if needed. :)

jamesmunns, 2019-02-04:
I'll be there, by the way :)

I'd also like to nominate this issue to be closed, unless we plan to coordinate anything here.

hannobraun, 2019-02-04:
Oh, I forgot to update this issue: @mre and I have been accepted and will be doing a workshop.

I agree that we can close this issue. I think it served its purpose.

closed 2019-02-04 by hannobraun