rust-embedded/wg #239: Embedded WG meeting, October 16

japaric, 2018-10-16:
[8-9 PM CEST][0] on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2018-10-16,360,b8jj

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

Agenda.

- Triaging [edition issues](https://github.com/search?q=org%3Arust-embedded++is%3Aopen+milestone%3A2018&type=Issues)

- [Issues nominated for discussion, if any](https://github.com/rust-lang-nursery/embedded-wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

thejpster, 2018-10-23:
I could see an issue for the meeting on 23 Oct, but FYI I may not be able to attend due to work stuff. I'll see what I can do.

japaric, 2018-10-23:
This meeting happened! The minutes can be found in #242.

The agenda for the next meeting is in #241.

@thejpster no worries if you can't make it

closed 2018-10-23 by japaric