rust-embedded/wg #245: Embedded WG meeting, October 30

japaric, 2018-10-25:
[8-9 PM CET][0] on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2018-10-30,420,b8jj

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

Agenda.

- Triaging [edition issues](https://github.com/search?q=org%3Arust-embedded++is%3Aopen+milestone%3A2018&type=Issues)

- [Issues nominated for discussion, if any](https://github.com/rust-lang-nursery/embedded-wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

japaric, 2018-10-25:
Reminder that our meetings are at 8 PM *Berlin* time and that (most of?) Europe is doing that weird DST thing on Sunday. So double check the time next week so that you are not earlier by an hour, or late by an hour, or something like that.

hannobraun, 2018-10-25:
I won't be attending this time, as the meeting conflicts with a local event I'd like to visit.

jamesmunns, 2018-10-30:
I will likely not be able to attend. Here are my notes for this week:

* Regarding the Portability chapter, @therealprof drafted (and we've merged) a first pass in https://github.com/rust-embedded/book/pull/64
    * There are some follow up to-dos in https://github.com/rust-embedded/book/issues/65
* Regarding "frozen" links for the new Rust webpage, I have some answers and one open question in https://github.com/rust-embedded/book/issues/52: the question is whether to use `bookshelf` or `book` links for the website. If we have an answer during the meeting, I will make the changes (including to the website) tomorrow.
* I wanted to bring https://github.com/rust-embedded/svd2rust/issues/213 up for discussion, but it seems like neither @hannobraun nor I will be present. If anyone wants to take a peek and comment, it would be appreciated. It can wait for next week.
* I went with a new format for the newsletter, as we only had a few shares over the last 3 weeks, and I wanted to change things up. Interested to hear feedback from others. [check it out here](https://rust-embedded.github.io/blog/2018-10-28-newsletter-14/)

therealprof, 2018-10-30:
I'd like to see whether we can come up with a plan to address the annoying CI fails on svd2rust.

jamesmunns, 2018-10-30:
@therealprof are they mostly transient network issues?

If so, I would suggest that we look at moving the travis testing to use the `svd2rust_regress` tool, rather than the current `bash`-based infra, and start adding retry logic there. We could also address this in the bash script, but I know at least personally, I am more comfortable in Rust than in bash.

jamesmunns, 2018-10-30:
Follow up from IRC, @therealprof was mentioning the failures that were likely due to incompatibilities with proc macros and rust `1.29.x`. It seems, at least so far, that this issue may go away now that testing is using `1.30.x`, where proc macros have stabilized.

japaric, 2018-11-06:
This meeting happened! The minutes can be found in PR #248 .

The agenda for the next meeting is in #249.

closed 2018-11-06 by japaric