rust-embedded/wg #246: gd32 support

puzrin, 2018-11-03:
- https://lcsc.com/search?q=gd32f3 (sort by price)
- http://www.gigadevice.com/products/microcontrollers/gd32/ (see selection guide pdf)

While exact clones of stm32 are not anything special (~ similar price), Giga Devices has "value line" - GD32F330 & GD32F350. To describe short - "similar to F0, but with floating point (!)".

It may be interesting to support those out of box, but i'm not sure is it possible to integrate support into existing HAL stm32 drivers.

AcrossTheWall, 2019-07-28:
In compiler control string, it is Cortex-M4, not Cortex-M4.fpu,

[image]

Dirbaio, 2024-06-11:
Closing this as part of 2024 triage.

As of now, the charter of the WG is to focus on foundational crates and functionality, rather than develop solutions for all use cases. However, as of today there are the following existing solutions, and folks finding this issue are suggested to contribute to one of the following projects:

- https://github.com/gd32-rust

If you think this was closed incorrectly, please leave a comment and we can revisit this decision in the next weekly WG meeting on Matrix.

closed 2024-06-11 by Dirbaio