rust-embedded/wg #249: Embedded WG meeting, November 6

japaric, 2018-11-06:
[8-9 PM CET][0] on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2018-11-06,420,b8jj

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

Agenda.

- Triaging [edition issues](https://github.com/search?q=org%3Arust-embedded++is%3Aopen+milestone%3A2018&type=Issues)

- [Issues nominated for discussion, if any](https://github.com/rust-lang-nursery/embedded-wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

hannobraun, 2018-11-08:
Sorry, something unexpected came up. I won't be making it to next week's meeting.

japaric, 2018-11-13:
This meeting happened! The logs are in PR #252.

The agenda for the next meeting is in #253

closed 2018-11-13 by japaric