rust-embedded/wg #256: Survey: 2019 wishlist

japaric, 2018-11-13:
Hello embedded Rustaceans!

We are collecting a "wishlist" of things you'd like to see done, fixed or
stabilized in 2019. We'll use this data to make a roadmap for 2019, i.e. to set
the goals of the WG for 2019.

You can request anything related to embedded Rust development: language
features, bug fixes, crates, docs, etc. The more requested something is the more
likely it is to make it into the roadmap; of course, other factors will be
considered as well: amount of work required; how beneficial would it be to the
community; does it require a rust-lang/rust RFC? how likely is that RFC to be
accepted?; who has to do the bulk of the work? the embedded community or a Rust
team?; etc.

To keep things tidy let's limit comments to **one** request per comment. Let's
also avoid "+1" and "I'd like to see that too" comments; instead use a :+1:
[reaction] to vote for a request.

[reaction]: https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/

To make our job easier please use the following format:

``` text
Category: pick one of:

- "feature", for languange and compiler features, e.g. const generics;
- "bug fix", e.g. fix rust-lang/cargo#5730;
- "stabilization", e.g. stabilize `MaybeUninit`;
- "crate", e.g. write / maintain a USB stack;
- "resources", for requesting docs, user guides, cookbooks and the like;
- or leave it empty if it doesn't fit any of the categories above

Request: your request in a single sentence

Rationale: one paragraph, or several bullets, explaining how this change would
benefit the embedded community
```

If you'd like to extend the rationale of an already submitted request just make
a new comment referencing the original request. We'll merge your comment into
the request comment. For example:

``` text
> Request: the sentence that Alice wrote

This would also let us do X and Y, and it would let crate Z compile on stable!
```

### Summary of requests

**TODO** add links to submitted requests; group the requests in categories

cc @rust-embedded/all please check this thread every now and then and update
the summary section

japaric, 2018-11-13:
**Category**: Stabilization

**Request**: Stabilize [RFC 2282][rfc] - "Cargo profile dependencies" AKA `profile-overrides`

[rfc]: https://github.com/rust-lang/rust/issues/48683

**Rationale**:

- This feature is used to optimize dependencies when compiling a crate using the dev profile; this produces a small binary that is still somewhat debuggable (`print x` doesn't always print `x = <optimized out>`). Without this sometimes dev binaries are too big and won't fit in the target device!

- This is also used to optimize specific crates, like `r0`, for size and mitigate the "bloat" of aggressive loop unrolling that LLVM does when the release profile (`-C opt-level = 3`) is used.

japaric, 2018-11-13:
**Category**: Bug fix

**Request**: fix rust-lang/cargo#5730 - "Features of dependencies are enabled if they're enabled in build-dependencies; breaks no_std libs"

**Rationale**: this makes some `no_std` crates unnecessarily hard to use / write. Using a crate, like `chrono` or `rand`, which supports both `no_std` and no-`no_std` contexts, in a build script or procedural will break the compilation of `no_std` crates that depend on the build dependency. The workaround is to use the host dependency in `no_std` mode everywhere, but this makes build scripts and procedural macros harder to write.

japaric, 2018-11-13:
**Category**: Stabilization

**Request**: Stabilize [`core::mem::MaybeUninit`][MU]

[MU]: https://doc.rust-lang.org/nightly/core/mem/union.MaybeUninit.html

**Rationale**: this is required to make [`heapless`]' (\*) and similar libraries' `const` constructors work on stable.

[`heapless`]: https://crates.io/crates/heapless

(\*) `heapless` is a widely used library that provides statically allocated, fixed capacity collections (`Vec`, `String`, `HashMap`, etc.).

japaric, 2018-11-13:
**Category**: Stabilization

**Request**: Stabilize `const fn`-s that have trait bounds, e.g. `const fn new() -> Mutex<T> where T: Send`.

**Rationale**: this is required to make [`heapless`]' (\*) and similar libraries' `const` constructors work on stable.

[`heapless`]: https://crates.io/crates/heapless

(\*) `heapless` is a widely used library that provides statically allocated, fixed capacity collections (`Vec`, `String`, `HashMap`, etc.).

ZirconiumX, 2018-11-14:
**Category**: "feature"

**Request**: MIPS intrinsic for `mfc0`/`mtc0`

**Rationale**: Access to the system control coprocessor in MIPS provides access for features ranging from MMU control (for devices that have it) to enabling/disabling interrupts and handling interrupts. This currently requires either external assembly, which is not optimal, or `asm!` which is highly unstable.

adamgreig, 2018-11-14:
**Category**: Stabilization

**Request**: thumb intrinsics (#63)

**Rationale**: many (all?) the intrinsics we want are in stdsimd (https://github.com/rust-lang-nursery/stdsimd/pull/518) and so available to nightly; it would be great to get them stabilised as well. The last RFC about this (#184) sort of stalled out.

P.S. thanks @ZirconiumX for reminding me!

ZirconiumX, 2018-11-14:
**Category**: None of the above

**Request**: Optimise memory routines in `compiler-builtins`.

**Rationale**: At the moment the `compiler-builtins` code is written to be correct and to make it as obvious for LLVM to optimise as possible. However, if LLVM does not have an internal routine to optimise these routines, it will emit the code directly, resulting in things like byte at a time `memcpy`, which is far from ideal. It's possible to write a 32-bit routine which should be portable to most machines and offer a ~4x speedup.

thejpster, 2018-11-14:
**Category**: Crate
**Request**: An easy to use SD/MMC and FAT32 library

**Rationale**: Adding SD card support to your embedded project should be as easy as using the Arduino [SD library](https://www.arduino.cc/en/Reference/SD) - just depend on the crate and add a on-liner connecting the SD card library to your chip's SPI peripheral. This library should prioritize ease of use over performance (i.e. polling and byte-wise transfers rather than interrupt-driven DMA block transfers), but allow for higher-performance implementations to be plugged in if required, through a generic `BlockDevice` trait.

thejpster, 2018-11-14:
*Category*: Crate
*Request*: An easy to use USB host library
*Rationale*: Adding support for USB thumb drives, keyboards and mice to your embedded project should be as simple as using the Arduino [USB Host](https://www.arduino.cc/en/Reference/USBHost) library.

thejpster, 2018-11-14:
**Category**: Crate
**Request**: An easy to use USB device library
**Rationale**: Adding support for emulating a USB keyboard, mice or serial port to your embedded project should be as simple as the two examples above ;)

thejpster, 2018-11-14:
**Category**: Crate
**Request**: An easy to use pure-Rust RTOS
**Rationale**: Binding to a pre-existing C based RTOS is non-trivial, as there are things the RTOS does (like memory allocation) which aren't necessarily a good fit for the Rust model. We should write a pure-Rust version of FreeRTOS, decomposed into re-usable core crates (queues, mutextes/semaphores, tasks, etc), and then microarch-specific implementations (e.g. rustos-queue-cortex-m).

eldruin, 2018-11-14:
**Category**: Bug fix
**Request**: Fix https://github.com/rust-lang/rust/issues/42863 - "Generic associated consts can't currently be used to parameterize fixed array lengths"

**Rationale**: In drivers that support several devices, it would be good to be able to define generic associated constants for things like buffer sizes, which and use them to create arrays

japaric, 2018-11-14:
**Category**: Feature

**Request**: std-aware Cargo / Xargo in Cargo

**Rationale**:

- with this feature people would be able to build `no_std` programs for custom targets, targets for which there is no `rust-std` component, using Cargo. The most prominent custom targets are consoles: Nintendo DS, GBA, PS2, etc. The workaround is to use a Cargo wrapper like Xargo or `cargo-xbuild` but these will always require nightly.

- this would also let people recompile core / std with different compilation settings, for example using `-C opt-level=s` instead of the default `-C opt-level=3`

This is planned work for the Cargo team so I mainly want to gauge interest and check how important is it for you to get this stabilized in 2019. Please vote with a :+1: if this would be great to have available in nightly, OR with a :tada: if it's vital for you to have this working on stable.

japaric, 2018-11-14:
**Category**: *unsorted*

**Request**: math support in core

**Rationale**: there's nothing OS specific about math libraries. `x.tan()` should Just Work in `no_std` context. There are `no_std` libraries that provide math support but having this in core and maintained by the Rust team means that we can freely use unstable features like inline assembly end SIMD to make the implementations as performant as possible.

DrSensor, 2018-11-14:
**Category**: Resource

**Request**: Table/List of comparison/example between embedded_hal API vs Arduino API

**Rationale**: This will help in teaching newcomer about embedded rust. Especially the one who doesn't have experience in firmware programming. Also, it makes it easier for someone to port some of Arduino libraries into rust drivers.

ramn, 2018-11-14:
**Category**: feature
**Request**: Support for AVR
**Rationale**: There are many Arduinos lying around. Would be nice to rust them up!

jed-frey, 2018-11-15:
- Category: feature
- Request: [Functional Safety Certification](https://en.wikipedia.org/wiki/Functional_safety), examples:
  - [US RTCA DO-178C North American Avionics Software](https://en.wikipedia.org/wiki/DO-178C)
  - [US RTCA DO-254 North American Avionics Hardware](https://en.wikipedia.org/wiki/DO-254)
  - [EUROCAE ED-12B European Airborne Flight Safety Systems]()
  - [IEC 62304 - Medical Device Software](https://en.wikipedia.org/wiki/IEC_62304)
  - [IEC 62061, Safety of machinery - Functional safety of safety-related electrical](https://en.wikipedia.org/wiki/IEC_62061)
  - [ISO 25119 - Tractors and Machinery for Agriculture and Forestry -- Safety-Related Parts of Control Systems](https://www.iso.org/standard/45050.html)
  - [ISO 26262 - Road Vehicles Functional Safety](https://en.wikipedia.org/wiki/ISO_26262)
- Rationale: People Dying = Bad.

jed-frey, 2018-11-15:
- Category: feature
- Request: [PowerPC e200](https://en.wikipedia.org/wiki/PowerPC_e200) core compiler support.
- Rationale: The NXP [MPC5748G](https://www.nxp.com/docs/en/data-sheet/MPC5748G.pdf) and [MPC5744P](https://www.nxp.com/docs/en/data-sheet/MPC5744P.pdf) are ASIL-B and ASIL-D certified chips, respectively. They feature various e200 cores. NXP also has 2 'inexpensive' (relatively speaking) development boards for each of the chips:
  - [DEVKIT-MPC5748G](https://www.nxp.com/products/processors-and-microcontrollers/power-architecture-processors/mpc5xxx-55xx-32-bit-mcus/ultra-reliable-mpc57xx-32-bit-automotive-and-industrial-microcontrollers-mcus/development-board-for-mpc5748g:DEVKIT-MPC5748G) - *Ultra-Reliable MCUs for Automotive & Industrial Control and Gateway*
  - [DEVKIT-MPC5744P](https://www.nxp.com/products/processors-and-microcontrollers/power-architecture-processors/mpc5xxx-55xx-32-bit-mcus/ultra-reliable-mpc57xx-32-bit-automotive-and-industrial-microcontrollers-mcus/development-board-for-mpc5744p:DEVKIT-MPC5744P) - *dual e200z4 lockstep cores, motor control, safety, and communication interfaces to facilitate a complete safety/chassis solution.*

Sh4rK, 2018-11-15:
**Category**: feature
**Request**: Stabilize async/await (and all needed underlying machinery).
**Rationale**: It's great that embedded-hal supports async via the [nb](https://crates.io/crates/nb) crate, but if I'm not mistaken, most drivers and user code use the blocking interface. It would be great if they could transition to being asynchronous, to be used in a multitasking system.

Sh4rK, 2018-11-15:
**Category**: crate
**Request**: A task scheduler.
**Rationale**: Once we have async support (https://github.com/rust-embedded/wg/issues/256#issuecomment-438929923), we should have a task scheduler that makes it possible to write multi-tasked embedded programs. In this context what I mean by a task would probably be a Future from the core library. I'm not exactly sure on the specifics, but I'm thinking something like [rtfm](https://github.com/japaric/cortex-m-rtfm) (compile-time guarantee of safe and efficient access to shared resources), but with the priorities assigned to tasks (Futures), instead of specific interrupts. A possible option is to actually implement this in rtfm itself.

(Side note: I would probably be able to help writing this.)

paoloteti, 2018-11-15:
**Category**: feature

**Request**: Add support for Cortex-R52

**Rationale**: One of the most advanced real-time embedded CPU out there. As the first `Armv8-R` processor, Cortex-R52 introduces support for an hypervisor. It is part of the NXP S32 automotive platform.

grossws, 2018-11-15:
**Category**: stabilization

**Request**: stabilize `alloc`-related functional (`alloc` and `allocator_api`), see [RFC 1398](https://github.com/rust-lang/rfcs/blob/master/text/1398-kinds-of-allocators.md) and related issues/rfcs

**Rationale**: It would allow to use dynamic heap-allocated structures on stable

jacobrosenthal, 2018-11-15:
**Category**: resources

**Request**: are-we-no-std-yet style community push

**Rationale**: push no_std amongst important libraries and a guide to helping library authors avoid std stuff they may not need, feature gate stuff they do

geomatsi, 2018-11-16:
**Category**: resources

**Request**: Improve documentation needed to start experimenting with Rust on AVR, MSP430, RISC-V

**Rationale**: The more people start to experiment with those platforms the better - more testing, more drivers, etc. Using Rust on cortex-m chips has been greatly simplified and streamlined during this year. Other mcu platforms are not yet there and have a more steep learning curve now. It would be great to have introductory documents for some available development boards (LaunchPads, Arduinos) explaining how to get started with Rust on those boards: how to prepare environment (Xargo/Cargo), maintained crates (HAL, drivers, etc), troubleshooting (e.g. where to post LLVM issues), and so on.

therealprof, 2018-11-17:
**Category**: resources

**Request**: Improve crates.io or create separate site for `no_std` crates

**Rationale**: It is still very hard to find suitable crates for `no_std` development on crates.io since searching for keywords is hit and miss and it's not possible to include/exclude categories.

therealprof, 2018-11-17:
**Category**: resources

**Request**: Create site listing all available `embedded-hal` implementations with supported MCUs and peripherals

**Rationale**: We now have a sizeable amount of HAL impls with widely varying support of various peripherals and some even supporting multiple MCUs. It would be nice to have an overview of what's already supported (to help selecting a suitable MCU) or where a hand is needed to improve support.

therealprof, 2018-11-18:
**Category**: feature

**Request**: Language support for types with storage size of less than 1 byte

**Rationale**: Currently in Rust the smallest available non-ZST is 1 byte which means that arrays of `bool` or small but non-ZST types in structs will require more room than in other languages and reduce interface compatibility. It also requires PACs to emulate access to specific parts of a memory mapped register thereby massively inflating source code size and compile times and reducing readability. There's some RFC in the works already but that could use some more love: https://github.com/rust-lang/rfcs/pull/2581

ketsuban, 2018-11-18:
**Category**: Feature

**Request**: Add `a32` and `t32` features to `target_feature` on ARM targets, mapping to the LLVM features `-thumb-mode` and `+thumb-mode`

**Rationale**: On the Game Boy Advance while most code uses the Thumb ISA for performance reasons (the cartridge is on a sixteen-bit bus, so ARM instructions incur a penalty) some things (e.g. an interrupt dispatch function, should you choose to provide one) need to be ARM. It would be desirable to write these function in Rust code rather than raw assembly, but `thumb-mode` is not currently whitelisted for `target_feature`. I initially suggested [`#[arm]`/`#[thumb]` function attributes](https://internals.rust-lang.org/t/pre-proposal-arm-thumb-function-annotations-for-arm-targets/8816), but it was pointed out to me that [`target_feature` features no longer have to map simply onto LLVM ones](https://internals.rust-lang.org/t/pre-proposal-arm-thumb-function-annotations-for-arm-targets/8816/4).

TotalKrill, 2018-11-18:
**Category**:  Resources

**Request**:  Recommended/Supported/Blessed way of doing unit or integration tests on Embedded systems.

**Rationale**: Bringing modern, well tested development methods to the embedded space would benefit us all. I saw no mention of this in the Embedded book. Embedded testing also differs from regular unit testing since it most often will include hardware in the loop.

korken89, 2018-11-18:
**Category:** bug fix

**Request:** Fix so infinite loops (`loop {}`) does not need an atomic compiler fence to not be optimized into an abort instruction

**Rationale:** This is very un-intuitive for a C/C++ user coming to Rust and should not happen. This was under discussion before, but I am not able to find the related issue, maybe someone knows which one it was?

roblabla, 2018-11-18:
@korken89 https://github.com/rust-lang/rust/issues/28728?

adamgreig, 2018-11-18:
I have added all requests up to this point to the issue description.

roblabla, 2018-11-19:
**Category:** feature request

**Request:** `std::io` in libcore/liballoc

**Rationale:** Most of `std::io` is fully platform independent (The traits and Cursor), and are very very useful abstractions to use. Many crates rely on std::io::Cursor to parse byte arrays, but don't otherwise rely on the platform-specific bits.

albru123, 2018-11-19:
**Category**: resources

**Request**: Complete usage examples for each trait in `embedded-hal`

**Rationale**: Currently it is quite hard to find resources on the `embedded-hal` crate usage, because the examples and implementation details are scattered between `embedded-hal`, `stm32f30x-hal` (or any other implementation crate) and the `f3` where the usage examples are present. It would be better to have all the necessary stuff for beginners in one place so it's easy to pick up.

Lokathor, 2018-11-20:
**Category**: feature

**Request**: ability to assert within a `const fn` context, with a compilation error if the assertion fails at compile time and a panic if the assertion fails at runtime.

**Rationale**: there are often known limits of input (eg "must be non zero", "must be less than 32", etc) that we would like to codify into our `const` functions and have work smoothly regardless of if the function is used at compile time or run time. Currently you cannot use `if` in a `const` context, so you cannot assert. There is the [static assertions](https://docs.rs/static_assertions/) crate, but that _only_ works in a compilation context, it doesn't also work at runtime. Of course, increasing what can be done with `const fn` is already on the general Rust roadmap; I suppose I'm just advocating that the wg help speed this particular element along if possible.

japaric, 2018-11-20:
I have added all requests up to this point to the issue description.

I have also created a new category: "Std" for addition and changes to any of the crates that are part of the standard library, i.e. core, alloc, compiler-builtins, etc. I have moved the two "unsorted" requests into this category; I have also moved "std::io in libcore/liballoc" into this new category.

japaric, 2018-11-30:
**Category**: feature

**Request**: more control over loop unrolling (See `#pragma unroll` and `-funroll-loops`)

**Rationale**: LLVM does aggressive loop unrolling when opt-level is equal or greater than 2. On Cortex-M loop unrolling can speed up some routines by a factor of 2 at the cost of hundreds of extra bytes of .text (Flash). For devices with limited amount of Flash one would like to avoid loop unrolling; currently the only choice is to optimize for size (`opt-level={s,z}`) but this penalizes other optimizations due to their lower inline threshold. With more compiler knobs for loop unrolling it would be possible to use `opt-level=3` while globally disabling loop unrolling and manually enabling it for loops where the speed - binary size tradeoff is worthwhile.

japaric, 2018-11-30:
**Category**: feature

**Request**: ship GDB compiled with --enable-targets=all with the toolchain

**Rationale**: To debug a remote target one usually needs to install an *extra* GDB that understands the target architecture (e.g. `arm-none-eabi-gdb`). By shipping a GDB with multi-arch support with the toolchain installing a *single* GDB that supports the host (x86_64), ARM and other architectures would be as simple as running `rustup component add rust-gdb`.

puzrin, 2018-12-01:
**Category**: crates

**Request**: GUI with desktop simulation support

**Rationale**: Device interface development may be the most time consuming process. In the same time, the most of things can be quickly prototyped on PC, without hardware. Need something like [littlevgl](https://github.com/littlevgl/lvgl).

daxpedda, 2018-12-01:
**Category**: feature

**Request**: Support defining enabled and disabled lints in a configuration file (https://github.com/rust-lang/cargo/issues/5034, https://github.com/rust-lang/cargo/pull/5728)

**Rationale**:
> Currently project-wide lint configuration needs to be included in the crate sources. This is okay when the project consists of a single crate, but gets more cumbersome when the project is a workspace with dozen crates.
> 
> Being able to define the lints in an external file that will be used when building the crates would have several benefits:
> 
> * Sharing one configuration file for all the crates in a workspace.
> * A canonical place to check for lint configuration when contributing to a new project.
> * An easy way to examine the version history for lint changes.

little-arhat, 2018-12-05:
@albru123 

"Complete usage examples for each trait in embedded-hal" -- indeed! 

We have number of examples in the [our proving-ground repo](https://github.com/copterust/proving-ground), but it would be nice to have smth standard.

eldruin, 2018-12-06:
**Category**: bug fix

**Request**: Fix https://github.com/rust-lang/rust/issues/54973

**Rationale**: This bug is the reason why the ADC trait API is not as nice as it could be. See [here](https://github.com/rust-embedded/embedded-hal/issues/110#issuecomment-437688937)

monouser7dig, 2018-12-07:
**Catergory** resources

**Request** add end to end workflow for popular microcontroller such as arduinos

**Rationale**
- [ ] a JTAG like **debugger experience**, preferably by using another Arduino? I've seen sth. like this being done but can not do it yet but this is really important, in the end **gdb should be attached to the real hardware**
- [ ] just basic instructions everyone can follow and have a blinking LED at the end  
  - [ ] installing the right rust component
  - [ ] setting up ports on the pc
  - [ ] compiling
  - [ ] uploading
- [ ] if Arduino is not possible due to llvm issues choose another very popular microcontrollers but Arduinos are the elegant in the room for enthusiasts

TeXitoi, 2018-12-07:
@monouser7dig https://github.com/TeXitoi/blue-pill-quickstart should almost satisfy your request, except that it's not an arduino. This dev board is compatible with the arduino SDK.

monouser7dig, 2018-12-07:
@TeXitoi yes correct that looks about what I want just for more boards, thanks 👍

Lokathor, 2018-12-09:
**Category:** feature

**Request:** UnsafeDeref trait

**Rationale:** Sometimes you have a raw pointer (people often ask me why, doesn't matter why) and you need to access a field on a struct, `(*p).f` But sometimes that field is itself leading to another field, `(*(*p).f).z` and as you can see it gets horribly ugly very quickly. All I want is the ability to write `p.f` within an unsafe block and have it do the right thing.

sajattack, 2018-12-09:
**Category**: std

**Request**: Ability to define a fmt_implementation, similar to a panic_implementation

**Rationale**: We could provide an implementation that does nothing, saving large amounts of binary size when including crates or other parts of std that use fmt::Debug or fmt::Display by default.

https://github.com/rust-embedded/wg/issues/247

jamwaffles, 2018-12-09:
**Category**: std

**Request**: Ability to use `format!()` in non-std (embedded) contexts

**Rationale**: [embedded-graphics](https://crates.io/crates/embedded-graphics) and other libraries that require input/output of formatted text would benefit greatly from the ability for either library code or application code to use `format!()`. For example, printing an accelerometer value to an OLED display using `display.draw(Font6x7::render_str(format!("Temp: {}", the_temp));`.

AxFaure, 2018-12-09:
**Category**: feature

**Request**: ability to list/forbid dependencies using unsafe code or which could panic

**Rationale**: In order to guarantee a minimum level of safety/security in our code, we need to be able to determine to which extent we can trust the libraries we are using. Having the option to forbid unsafe code and/or code that can panic in our crate and in its dependencies would be a huge plus. Since not all unsafe code can be avoided, it could probably be implemented as a whitelist that could be configured in cargo (or maybe as a plugin similar to clippy). A developer could then allow some unsafe dependencies after manually reviewing them. This is probably not specific to the embedded world but there is a lot of embedded software that is critical enough to need this feature. There are some crates like this one https://crates.io/crates/cargo-geiger that can already help but I think more work is needed in this direction

japaric, 2018-12-09:
I have added all requests up to this point to the issue description.

TeXitoi, 2018-12-10:
The `write!` macro works on no_std. I can't see how the `format!` macro can work in a no_std context.

puzrin, 2018-12-10:
**Category:** crates (?)

**Request:** Easy firmware flashing for popular OS-es / SOCs (initially - stlink/DFU for linux/osx/win).

**Rationale:** [Original](https://github.com/rust-embedded/book/issues/12#issuecomment-411958975) issue is now in book repo. But since it can not be resolved by documentation only, i think it worth place it here.

Let's imagine, you developed great OSS hardware, and users wish to repeat it or join to further development. Then it's critical to resolve 2 use cases:

- Users (non programmers) should be able flash firmware without reading docs at all. They just need big button "do everything perfect" :)
- The same for potential collaborators. When people wish to help you - no need to waste their time for automate-able things.

Right now, with C and PlatformIO i can give very simple instructions:

1. Install vscode, open project and wait until it downloads all necessary tools for your OS.
2. Attach programmer and select "upload" from menu.

I understand, this problem is not Rust-specific. But since Rust plans to be awesome for embedded, it's important to be awesome about firmware upload too.

aurabindo, 2018-12-12:
**Category**: Resources/None

**Request**: Guidelines for vendor participation

**Rationale**: If there is an official point of contact and established procedure for vendors to submit drivers for their devices, ecosystem will flourish. I'm proposing a model equivalent to the one followed by Linux. Vendor or independent contributors submits code, gets scrutinized in the public, and gets accepted into an official source controlled by the WG. Advantages:

1. Ecosystem will have high quality code, and it becomes easy for a starter to look up the current state. With no/few officially endorsed crates (device specific hal implementations), users will have to do more research to find the optimal solution.
2. More vendor backing => increased funding/chances of certifications. Rust is certainly miles ahead compared to C as a tool for building safety critical systems. In real world, organizations still need certifications from various authorities. It is a long road, but we can start, I hope!

(I posted about this before coming across this issue here: https://github.com/rust-embedded/wg/issues/264)

eddyp, 2018-12-14:
>     * Request: [PowerPC e200](https://en.wikipedia.org/wiki/PowerPC_e200) core compiler support.
>       The NXP [MPC5748G](https://www.nxp.com/docs/en/data-sheet/MPC5748G.pdf) and [MPC5744P](https://www.nxp.com/docs/en/data-sheet/MPC5744P.pdf) are ASIL-B and ASIL-D certified chips, respectively. They feature various e200 cores. NXP also has 2 'inexpensive' (relatively speaking) development boards for each of the chips:

To support NXP platfroms with e200 cores, llvm will need to support the code-dense variation of the powerpc instruction set, VLE, because MPC57xxx MCUs are VLE only, but it seems there is no support for this, athough there have been some experiments (amybe just behind closed doors):
http://lists.llvm.org/pipermail/llvm-dev/2014-July/074613.html

avraliov, 2018-12-17:
**Category**: feature

**Request**: bits layout implementation like in Ada

**Rationale**: in Ada we can describe bits layout like:
Some_Record is record with
      field1 : integer;
      field2 : integer;
end record;
for Some_Record use record
     field1 at 0 range 0..5; //bits layout
     field2 at 0 range 6..7; //bits layout
end record;

Will be nice to have something like that declarative stile in Rust embedded

eddyp, 2018-12-17:
@avraliov I would rather have something less spread out like:

struct SomeRecord {
    reg1:u8 bits{
       f1:0..5,
       f2:6..7,
    },
    reg2:u32
}

Or something maybe less verbose like:
struct SomeRecord {
    f1:bits 0..5 in reg1:u8,
    f2: bits 6..7 in reg1:u8,
    f3: bits 4..17 in reg2:u32,
    }
although I kind of dislike the repetition of the "in reg1:u8" part.

Lokathor, 2018-12-20:
**Category:** Feature

**Request:** Make `example-dependencies` and `test-dependencies` be split up, instead of having both as `dev-dependencies`

**Rationale:** Tests usually want quickcheck or other std assuming crates, examples usually want to be no_std examples that demonstrate the use of the crate. Right now there's no way to handle this.

eddyp, 2018-12-20:
@Lokathor For unit tests, you could run them on a different architecture than the target, I do that in C. Being able to do that same thing in Rust would be awesome.

TeXitoi, 2018-12-20:
@eddyp That's doable https://github.com/TeXitoi/rusty-clock/blob/master/.travis.yml

eddyp, 2018-12-20:
@TeXitoi OK, looks like what I wanted, but doesn't that explicitly require to mark in the Rust code which code is also runnable/testable on x86_64? I'm thinking the dependencies on arch specific crates such as cortex-m might make the generated code unusable otherwise.

[16 more replies]

closed 2020-09-01 by adamgreig