rust-embedded/wg #259: Add code snippets to embedded web page?

japaric, 2018-11-30:
The new Rust website had a [beta launch](http://beta.rust-lang.org/) yesterday :tada: . Here's the [embedded webpage](http://beta.rust-lang.org/what/embedded).

We have the option of adding code samples to the embedded web page (before the year ends). For an example of how these would look check out the [CLI webpage](http://beta.rust-lang.org/what/cli).

This issue is to discuss what code snippet we should add (if any!). I think that due to space constraints the snippets should be short and to the point.

Once we have made a decision we can report it back to rust-lang/beta.rust-lang.org#350

cc @rust-embedded/cortex-m @rust-embedded/resources

japaric, 2018-11-30:
Copy-pasting @jamesmunns' comment:

For the embedded WG, I would suggest one of the examples from https://docs.rust-embedded.org/book/start/registers.html, in particular I'm fond of this example, while a bit long, is a pretty good "full, standalone example":

``` rust
#![no_std]
#![no_main]

extern crate panic_halt; // panic handler

use cortex_m_rt::entry;
use tm4c123x_hal as hal;
use tm4c123x_hal::prelude::*;
use tm4c123x_hal::serial::{NewlineMode, Serial};
use tm4c123x_hal::sysctl;

#[entry]
fn main() -> ! {
    let p = hal::Peripherals::take().unwrap();
    let cp = hal::CorePeripherals::take().unwrap();

    // Wrap up the SYSCTL struct into an object with a higher-layer API
    let mut sc = p.SYSCTL.constrain();
    // Pick our oscillation settings
    sc.clock_setup.oscillator = sysctl::Oscillator::Main(
        sysctl::CrystalFrequency::_16mhz,
        sysctl::SystemClock::UsePll(sysctl::PllOutputFrequency::_80_00mhz),
    );
    // Configure the PLL with those settings
    let clocks = sc.clock_setup.freeze();

    // Wrap up the GPIO_PORTA struct into an object with a higher-layer API.
    // Note it needs to borrow `sc.power_control` so it can power up the GPIO
    // peripheral automatically.
    let mut porta = p.GPIO_PORTA.split(&sc.power_control);

    // Activate the UART.
    let uart = Serial::uart0(
        p.UART0,
        // The transmit pin
        porta
            .pa1
            .into_af_push_pull::<hal::gpio::AF1>(&mut porta.control),
        // The receive pin
        porta
            .pa0
            .into_af_push_pull::<hal::gpio::AF1>(&mut porta.control),
        // No RTS or CTS required
        (),
        (),
        // The baud rate
        115200_u32.bps(),
        // Output handling
        NewlineMode::SwapLFtoCRLF,
        // We need the clock rates to calculate the baud rate divisors
        &clocks,
        // We need this to power up the UART peripheral
        &sc.power_control,
    );

    loop {
        writeln!(uart, "Hello, World!\r\n").unwrap();
    }
}
```

japaric, 2018-11-30:
Copy-pasting another comment from @jamesmunns:

@therealprof suggested this blinky light example, which would be easier to trim down to size:

https://github.com/therealprof/stm32f042-hal/blob/master/examples/blinky_delay.rs

japaric, 2018-11-30:
I'm particularly fond of the `cortex-m-rt` attributes and safe `static mut` variables so I would suggest an snippet that shows that you can write code similar to the one you would write in C but in a way that's guaranteed to be memory safe:

``` rust
use cortex_m_rt::exception;
use cortex_m_semihosting::hprintln;

// declare an exception handler
#[exception]
fn SysTick() {
    // safe static variable
    static mut COUNT: u32 = 0;

    *COUNT += 1;

    // print to the host console
    hprintln!("SysTick: called {} times", *COUNT);
}
```

I'm not sure what a good caption would be, though.

therealprof, 2018-12-04:
@thejpster Exchanging data with other handlers/main is a separate topic IMHO and deserves it's own example. There're still uses for static data in interrupt handlers, e.g. to synthesise a time dependent signal (PWM or waveform).

TeXitoi, 2018-12-05:
I think a blinking led is really natural as such an example.

The code should be as small as possible. @therealprof blink example is great. Some idea to simplify:
 - using match instead of unwrap may be overkill, panicking in this context seems better, maybe using `expect()`? It will simplify indentation and the strange hacky `loop { continue; }`
 - `led.toggle()` can remove 2 lines in the main loop?
 - Maybe some lines can be merged to mask the complexity of the peripheral handling

```rust
let mut delay = Delay::new(cp.SYST, p.RCC.constrain().cfgr.freeze());
```

Ideally, a blink program using exception handler would be great, but I don't think we can do that without unsafe, and I think the example should be unsafe free.

TeXitoi, 2018-12-05:
Or a PWM driven blinking led?

flundstrom2, 2018-12-05:
I think both Blinky and the SysTick exception handler suggested by @japaric are very good examples. The former mainly because it shows how to access IO registers through the HAL, and the latter to show how easy it is to implement a handler.

LeonardMH, 2018-12-07:
Blinky, being the embedded version of hello world, is a natural example for a general audience like what you're going to get on the main rust-lang website.

I also agree with the suggestions @TeXitoi made above of using unwrap/expect rather than `loop { continue; }` (which feels like an anti-pattern in this case) and using led.toggle().

therealprof, 2018-12-07:
@TeXitoi Well this particular example uses `Delay`, I do have more versions, like one that loops (https://github.com/therealprof/stm32f042-hal/blob/master/examples/blinky.rs) and one which uses `SysTick` (no unsafe if you remove the suggested but optional initialisation of the SysTick counter value: https://github.com/therealprof/stm32f042-hal/blob/master/examples/flash_systick.rs).

`toggle` will only work if implemented and not with the two examples above. If we're aiming for a minimally viable example to blink a LED we could do:

```
#![no_main]
#![no_std]

extern crate cortex_m;
extern crate cortex_m_rt;
extern crate panic_halt;

extern crate stm32f042_hal as hal;

use hal::prelude::*;
use hal::stm32;

use cortex_m::asm::delay;
use cortex_m_rt::entry;

#[entry]
fn main() -> ! {
    let p = stm32::Peripherals::take().unwrap();
    let gpioa = p.GPIOA.split();
    let mut led = gpioa.pa1.into_push_pull_output();

    loop {
        led.set_high();
        delay(8_000_000);

        led.set_low();
        delay(8_000_000);
    }
}
```

(sans the external crate stuff for 2018 edition). Please note that this does not compile for me on the targeted platform ATM (https://github.com/rust-embedded/cortex-m/issues/125).

@LeonardMH 
> loop { continue; } (which feels like an anti-pattern in this case) and using led.toggle().

The example was trimmed for size but loops without side effects are frowned upon and create a lint, hence the `continue` the natural choice would be a simple `unreachable!()` which will add some additional bytes for the panic handler to the binary.

TeXitoi, 2018-12-10:
I've done an example that corresponds to everything that we talked here, that works on stable edition 2018. Open to review, especially around comments. Highly inspired by https://github.com/therealprof/stm32f042-hal/blob/master/examples/flash_systick.rs by @therealprof 

The static is a bit artificial.

https://github.com/TeXitoi/blue-pill-quickstart/blob/master/src/main.rs

japaric, 2018-12-11:
Space is at premium. AFAICT, the biggest snippet is 13 LoC and about 45 chars in width. How about blinking an LED from an exception handler?

*Stateful exception / interrupt handlers*

``` rust
use cortex_m_rt::exception;

// Blink an LED every second
#[exception]
fn SysTick() {
    static mut ON: bool = false;

    LED.set(*ON);
    *ON != *ON;
}
```

An LED is a reasonable thing to have in a `static` since it can allow mutation through a shared reference (see STM32's stateless ODR register)

japaric, 2018-12-11:
Another idea brought up during the meeting:

``` rust
#[entry]
fn main() -> ! {
    let spi: impl Spi = ..;
    let nss: impl DigitalOut = ..;
    let driver = Gyroscope::new(spi, nss); // moved
    // nsss.set_low(); // not allowed

    // ..
}
```

Showing a driver instantiation and move semantics.

adamgreig, 2018-12-11:
I'm in favour of a really short snippet, of a similar length to @japaric's example. I think we should try and show off one (maybe two) really nice features of embedded rust, without worrying about getting every detail right or having all the boilerplate.

So skip `extern bla`, `use bla`, etc, and just have a clear example of one thing. The suggestion of showing move semantics is nice -- perhaps demonstrating that a pin is taken as a singleton once, can be transformed into a new variable with a new type (default input into a push-pull output etc), moved into a driver (e.g. the SPI use case above), and then not be usable after move.

Or some other feature - maybe just show off how easy it is to use a driver targeting embedded-hal with an implementation of embedded-hal, since that sort of interoperability is basically unheard between disjoint libraries in C.

adamgreig, 2018-12-11:
> Maybe something similar could be done for sample(s) on the embedded site to hide such boilerplate code?

That would be great! I think ideally I'd like to see an entire repository linked which contains not only the right main.rs but also .cargo/config, Cargo.toml, memory.x, and is all ready to build -- since all those other files are just as necessary (and boilerplate) as the preamble in main.rs itself.

jamesmunns, 2018-12-18:
We discussed two items in the IRC meeting. In general, we want to keep these snippets simple, and easy to mentally parse for those unfamiliar with Rust. We discussed having two examples, one that shows off the syntax/blinky lights, and the other that focuses on something more complex, like traits or ownership.

My suggestion for the first, would be to use something like @therealprof suggested above, but trimmed down to not be 100% compile ready, but still representative:

```rust
// Include packages, such as hardware abstraction libraries for microcontrollers
use stm32f042_hal::stm32;

#[entry]
fn main() -> ! {
    let p = stm32::Peripherals::take().unwrap();
    let gpioa = p.GPIOA.split();
    let mut led = gpioa.pa1.into_push_pull_output();

    loop {
        led.set_high();
        delay(8_000_000);

        led.set_low();
        delay(8_000_000);
    }
}
```

For the second one, I suggested just showing a trait definition to serve as a conversation point for the kinds of abstractions you can make with the help of Embedded HAL. Perhaps such as the SPI trait definition:

```rust
/// Full duplex (master mode)
pub trait FullDuplex<Word> {
    /// An enumeration of SPI errors
    type Error;

    /// Reads the word stored in the shift register
    fn read(&mut self) -> nb::Result<Word, Self::Error>;

    /// Sends a word to the slave
    fn send(&mut self, word: Word) -> nb::Result<(), Self::Error>;
}
```

therealprof, 2018-12-19:
@jamesmunns 

This can be further reduced/refined to:

```
// Include packages, such as hardware abstraction libraries for microcontrollers
use stm32f0xx_hal::stm32;

#[entry]
fn main() -> ! {
    let p = stm32::Peripherals::take().unwrap();
    let mut led = p.GPIOA.split().pa1.into_push_pull_output();

    loop {
        led.toggle();
        delay(8_000_000);
    }
}
```

eldruin, 2020-08-11:
The website design has changed so we should revisit this. Nominating.

therealprof, 2020-08-18:
Can we come up with some reasonable generic snippets?

closed 2021-03-30 by therealprof