rust-embedded/wg #273: Embedded Linux Team: Decide on 2019 objectives

adamgreig, 2018-12-11:
As part of our 2019 planning, we want to figure out what each team wants to work on this year, including new ideas, items from the 2019 wishlist (#256), and any leftover items from this year. 

Let's use this issue to come up with a list of possible items and then prioritise them.

japaric, 2019-01-09:
We discussed in yesterday's meeting (#290) that teams should start deciding on
how to prioritize their work items for this year. We'll let each team decide how
to do the item prioritization but we recommend having a real-time / synchronous
meeting (IRC or video meeting) to discuss all the potential goals.

If appropriate, you can reserve a time slot during the weekly meetings to have
this synchronous discussion. Just let us know in advance in one of the meeting
issues (e.g. #291).

Finally, if any of your goals for this year requires changes in any of the
rust-lang/* repositories and you haven't mentioned them in #256 or #269 then
please report them in #269 ASAP. We'll discuss prioritization of rust-lang
request during the next meeting (2019-01-15).

cc @rust-embedded/embedded-linux

eldruin, 2020-08-11:
2019 finished. Maybe we can create a follow up issue for 2020. Nominating for next week's meeting.

adamgreig, 2020-09-01:
Closing as we're now well into 2020; if anything outstanding from this list is still relevant please add it to the [not-yet-awesome embedded rust](https://github.com/rust-embedded/not-yet-awesome-embedded-rust) list, or create a new issue to track it.

closed 2020-09-01 by adamgreig