rust-embedded/wg #274: HAL Team: Decide on 2019 objectives

adamgreig, 2018-12-11:
As part of our 2019 planning, we want to figure out what each team wants to work on this year, including new ideas, items from the 2019 wishlist (#256), and any leftover items from this year. 

Let's use this issue to come up with a list of possible items and then prioritise them.

japaric, 2018-12-13:
cc @rust-embedded/hal 

From the wishlist (from most voted to least voted):

- [An easy to use USB host library](https://github.com/rust-embedded/wg/issues/256#issuecomment-438569808)
- [An easy to use USB device library](https://github.com/rust-embedded/wg/issues/256#issuecomment-438570134)
- [Create site listing all available embedded-hal implementations with supported MCUs and peripherals](https://github.com/rust-embedded/wg/issues/256#issuecomment-439635403)
- [Complete usage examples for each trait in embedded-hal](https://github.com/rust-embedded/wg/issues/256#issuecomment-439827700)

Other ideas:

- Orgs like nrf-rs, stm32-rs and lpc-rs (or was it nxp-rs?) have been popping up. I think there's an opportunity there to coordinate and exchange notes (e.g. how should PACs be organized?) to provide a more unified experience to end users (e.g. do all provide a single vendor crate where device families are Cargo features?).

- This year we ran the weekly driver initiative which produced [over 60 driver crates](https://github.com/rust-embedded/awesome-embedded-rust#driver-crates). It would be interesting to ask the authors about their experiences: what worked? what was hard to do? This info could be used to improved `embedded-hal`. It would also be good to look for code duplication, or patterns, in the source code of the drivers; there may be things that can be abstracted into crates to simplify the implementation of drivers (e.g. I assume that both I2C and SPI driver crates deal with registers which authors have to create an API for by hand; can this process by simplified using macros or code generators?).

japaric, 2019-01-09:
We discussed in yesterday's meeting (#290) that teams should start deciding on
how to prioritize their work items for this year. We'll let each team decide how
to do the item prioritization but we recommend having a real-time / synchronous
meeting (IRC or video meeting) to discuss all the potential goals.

If appropriate, you can reserve a time slot during the weekly meetings to have
this synchronous discussion. Just let us know in advance in one of the meeting
issues (e.g. #291).

Finally, if any of your goals for this year requires changes in any of the
rust-lang/* repositories and you haven't mentioned them in #256 or #269 then
please report them in #269 ASAP. We'll discuss prioritization of rust-lang
request during the next meeting (2019-01-15).

cc @rust-embedded/hal

eldruin, 2020-08-11:
2019 finished. Maybe we can create a follow up issue for 2020. Nominating for next week's meeting.

adamgreig, 2020-09-01:
Closing as we're now well into 2020; if anything outstanding from this list is still relevant please add it to the [not-yet-awesome embedded rust](https://github.com/rust-embedded/not-yet-awesome-embedded-rust) list, or create a new issue to track it.

closed 2020-09-01 by adamgreig