rust-embedded/wg #275: MSP430 Team: Decide on 2019 objectives

adamgreig, 2018-12-11:
As part of our 2019 planning, we want to figure out what each team wants to work on this year, including new ideas, items from the 2019 wishlist (#256), and any leftover items from this year. 

Let's use this issue to come up with a list of possible items and then prioritise them.

japaric, 2018-12-13:
cc @rust-embedded/msp430 

From the wishlist:

- [Improve documentation needed to start experimenting with Rust on AVR, MSP430, RISC-V](https://github.com/rust-embedded/wg/issues/256#issuecomment-439326251)

Backlog:

- rust-lang/rust: `rust-std` component for `msp430-none-elf`. Then we drop the dependency on Xargo.
- bring msp430-rt up to parity with the latest cortex-m-rt. rust-embedded/msp430-rt#4. Then we can reduce the number of required unstable features to a single one.

Other ideas:

- LLVM seems to now have a proper object emitter / writer so we should move to it and stop depending on msp430-elf-gcc for object generation. (We'll still depend on it for linking though).

- We could pursue stabilizing the `msp430-interrupt` ABI.

japaric, 2019-01-09:
We discussed in yesterday's meeting (#290) that teams should start deciding on
how to prioritize their work items for this year. We'll let each team decide how
to do the item prioritization but we recommend having a real-time / synchronous
meeting (IRC or video meeting) to discuss all the potential goals.

If appropriate, you can reserve a time slot during the weekly meetings to have
this synchronous discussion. Just let us know in advance in one of the meeting
issues (e.g. #291).

Finally, if any of your goals for this year requires changes in any of the
rust-lang/* repositories and you haven't mentioned them in #256 or #269 then
please report them in #269 ASAP. We'll discuss prioritization of rust-lang
request during the next meeting (2019-01-15).

cc @rust-embedded/msp430

YuhanLiin, 2019-10-10:
This may be irrelevant, but how important is enabling debug info in MSP430 executables in the grad scheme of things? It would be nice to step through Rust source instead of ASM when debugging.

cr1901, 2019-10-10:
@YuhanLiin It's on my short-to-medium term list to add source-level debugging after the core crates (`msp430-rt`, etc) have been upgraded. That said, I've found w/ function sections enabled, ASM debugging is possible and not too horrible depending on the size of your MSP430.

YuhanLiin, 2019-10-10:
Do you know what steps would be involved for this feature? I may try my hand at it if I have time.

eldruin, 2020-08-11:
2019 finished. Maybe we can create a follow up issue for 2020. Nominating for next week's meeting.

adamgreig, 2020-09-01:
Closing as we're now well into 2020; if anything outstanding from this list is still relevant please add it to the [not-yet-awesome embedded rust](https://github.com/rust-embedded/not-yet-awesome-embedded-rust) list, or create a new issue to track it.

closed 2020-09-01 by adamgreig