rust-embedded/wg #276: RISC-V Team: Decide on 2019 objectives

adamgreig, 2018-12-11:
As part of our 2019 planning, we want to figure out what each team wants to work on this year, including new ideas, items from the 2019 wishlist (#256), and any leftover items from this year. 

Let's use this issue to come up with a list of possible items and then prioritise them.

japaric, 2018-12-13:
cc @rust-embedded/riscv 

From the wishlist:

- [Improve documentation needed to start experimenting with Rust on AVR, MSP430, RISC-V](https://github.com/rust-embedded/wg/issues/256#issuecomment-439326251)

Other thoughts:

- I think we should strive for making riscv-rt as close as possible to cortex-m-rt. That would let people carry their experience with one architecture to the other. It also means we can reuse the embedded book as introductory text instead of writing riscv specific docs.

- I think it might be possible to do RISCV development on stable, provided we adapt the existing crates. There are workarounds for (non inline) `asm!` and `global_asm!` on stable and there's no unstable, arch-specific ABI for RISCV, AFAIK.

dvc94ch, 2018-12-13:
As long as openocd is so buggy I can't imagine it being used seriously in commercial projects, the question is if it's easier to fix openocd or write a gdb stub + loader in rust.

Disasm, 2018-12-15:
I have developed a minimal runtime for picorv32 based on the ideas from cortex-m-rt and stable-embedded-rust. This runtime allows development on stable and works in hardware.
The reasons I have developed a separate runtime are the follows:
* picorv32 has specific IRQ handling, I don't have an idea how to implement this as part of riscv-rt
* I need a runtime which builds on stable (why not)
I am planning to implement proper IRQ handling in the near future. Ideally, I'd like to have something similar to cortex-m-rt.

https://github.com/Disasm/picorv32-rt
https://github.com/Disasm/picorv32-example

dvc94ch, 2018-12-23:
I can't speak for others on the team, but if I get around to doing some work for RISCV rust this year it would be getting gdb+ftdi+jtag+riscv-debug-spec working in pure rust. But I'm currently busy with new job and other stuff.

japaric, 2019-01-09:
We discussed in yesterday's meeting (#290) that teams should start deciding on
how to prioritize their work items for this year. We'll let each team decide how
to do the item prioritization but we recommend having a real-time / synchronous
meeting (IRC or video meeting) to discuss all the potential goals.

If appropriate, you can reserve a time slot during the weekly meetings to have
this synchronous discussion. Just let us know in advance in one of the meeting
issues (e.g. #291).

Finally, if any of your goals for this year requires changes in any of the
rust-lang/* repositories and you haven't mentioned them in #256 or #269 then
please report them in #269 ASAP. We'll discuss prioritization of rust-lang
request during the next meeting (2019-01-15).

cc @rust-embedded/riscv

japaric, 2019-02-06:
Triage:

@Disasm has done an awesome work with the riscv and riscv-rt crates and they now both compile on stable (see v0.4.x release). :tada: 

I have opened a PR (still WIP) to add an `entry` attribute to the riscv-rt crate: rust-embedded/riscv-rt#20

Disasm, 2019-02-15:
I'd like to work on the following:
* waiting for the RV64 support [to appear](https://github.com/rust-lang/rust/pull/58406)
* updating all the RISC-V crates (blocked on https://github.com/rust-embedded/svd2rust/pull/284 at the moment)
* adding picorv32 support
* adding HiFive Unleased support
* making cargo template for different targets: https://github.com/rust-embedded/wg/issues/146
* creating something like "cargo-flash" to make upload process easier

Disasm, 2019-04-02:
Triage:
RISC-V crates are updated and now work on stable. However, there is still work to be done.
At the moment I'm working on multi-core support.

eldruin, 2020-08-11:
2019 finished. Maybe we can create a follow up issue for 2020. Nominating for next week's meeting.

Disasm, 2020-08-18:
Triage:
* picorv32: @ilya-epifanov made a few crates for it
* K210 support: abandoned due to the lack of documentation, passed to @luojia65
* HiFive Unleashed support: abandoned, board is no longer in production
* cargo-flash: @Yatekii and @Tiwalun did a great job in probe-rs regarding this. Currently only the HiFive1 RevB board is supported (the main and still the most popular target for Rust on RISC-V).

We also made hal and pac libraries for GD32VF103 family.

luojia65, 2020-08-19:
We have a team continued on work for K210, its hal and how to make operating systems on it. Link: https://github.com/rcore-os/

adamgreig, 2020-09-01:
Thanks for the update!

I'm closing this issue since we're well past 2019; please feel free to make new issues for any further things that need tracking or add them to the [not-yet-awesome-embedded-rust list](https://github.com/rust-embedded/not-yet-awesome-embedded-rust).

closed 2020-09-01 by adamgreig