rust-embedded/wg #277: Resources Team: Decide on 2019 objectives

adamgreig, 2018-12-11:
As part of our 2019 planning, we want to figure out what each team wants to work on this year, including new ideas, items from the 2019 wishlist (#256), and any leftover items from this year. 

Let's use this issue to come up with a list of possible items and then prioritise them.

japaric, 2018-12-13:
cc @rust-embedded/resources 

From the wishlist (from most voted to least voted):

- [Recommended/Supported/Blessed way of doing unit or integration tests on Embedded systems.](https://github.com/rust-embedded/wg/issues/256#issuecomment-439686308)
- [are-we-no-std-yet style community push](https://github.com/rust-embedded/wg/issues/256#issuecomment-439229019)
- [Improve crates.io or create separate site for no_std crates](https://github.com/rust-embedded/wg/issues/256#issuecomment-439634901)
- [Improve documentation needed to start experimenting with Rust on AVR, MSP430, RISC-V](https://github.com/rust-embedded/wg/issues/256#issuecomment-439326251)
- [Create site listing all available embedded-hal implementations with supported MCUs and peripherals](https://github.com/rust-embedded/wg/issues/256#issuecomment-439635403)
- [Complete usage examples for each trait in embedded-hal](https://github.com/rust-embedded/wg/issues/256#issuecomment-439827700)
- [Table/List of comparison/example between embedded_hal API vs Arduino API](https://github.com/rust-embedded/wg/issues/256#issuecomment-438772936)
- [add end to end workflow for popular microcontroller such as arduinos](https://github.com/rust-embedded/wg/issues/256#issuecomment-445201443)

Book backlog:

- Guide for doing embedded development in an IDE. rust-embedded/book#12
- Profiling of embedded applications. rust-embedded/book#34
- Develop resources for Rust integration with other build systems. rust-embedded/book#61
- Develop resources for Rust integration with RTOSs. rust-embedded/book#62
- document #[interrupt]. rust-embedded/book#98
- HAL chapter: add examples. rust-embedded/book#99
- document how to use the ITM. rust-embedded/book#100
- memory safe DMA. rust-embedded/book#101

jamesmunns, 2019-01-08:
I'm interested in tackling [the testing topic](https://github.com/rust-embedded/wg/issues/256#issuecomment-439686308) - as I've had some thoughts around this.

I also think having [discoverability of embedded-hal impls](https://github.com/rust-embedded/wg/issues/256#issuecomment-439635403) is fairly important as well.

japaric, 2019-01-09:
We discussed in yesterday's meeting (#290) that teams should start deciding on
how to prioritize their work items for this year. We'll let each team decide how
to do the item prioritization but we recommend having a real-time / synchronous
meeting (IRC or video meeting) to discuss all the potential goals.

If appropriate, you can reserve a time slot during the weekly meetings to have
this synchronous discussion. Just let us know in advance in one of the meeting
issues (e.g. #291).

Finally, if any of your goals for this year requires changes in any of the
rust-lang/* repositories and you haven't mentioned them in #256 or #269 then
please report them in #269 ASAP. We'll discuss prioritization of rust-lang
request during the next meeting (2019-01-15).

cc @rust-embedded/resources

adamgreig, 2019-04-28:
Summary of WG meeting at Oxidize: things we'd like to work on this year:

* Rust book
    * How to use PACs
* Patterns book
* Link to training material
* Guide on how to write an embedded-hal driver (perhaps link to training material)

eldruin, 2020-08-11:
2019 finished. Maybe we can create a follow up issue for 2020. Nominating for next week's meeting.

adamgreig, 2020-09-01:
Closing as we're now well into 2020; if anything outstanding from this list is still relevant please add it to the [not-yet-awesome embedded rust](https://github.com/rust-embedded/not-yet-awesome-embedded-rust) list, or create a new issue to track it.

closed 2020-09-01 by adamgreig