rust-embedded/wg #280: Embedded WG meeting, December 18

japaric, 2018-12-17:
[8-9 PM CET][0] (Berlin time) on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2018-12-18,420,b8jj

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

Agenda.

- Code snippet for the embedded webpage? #259

- 2019 roadmap

- Should we check style with stable rustfmt in Travis?

- [Issues nominated for discussion, if any](https://github.com/rust-lang-nursery/embedded-wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

adamgreig, 2018-12-17:
Apologies, I won't be able to make this one. Have a good break all!

korken89, 2018-12-18:
I would like to get some input on #281 to help spread the word of Rust to C++ metaprogrammers

japaric, 2018-12-21:
This meeting happened! Minutes and IRC logs can be found in #283.

The next meeting is in **three weeks**. The agenda can be found in #284.

closed 2018-12-21 by japaric