rust-embedded/wg #286: Rust support for NeoTrellis (M4)

tarcieri, 2018-12-27:
Hello all. I'm looking for other people who might be interested in developing some Rust crates for use with the NeoTrellis and NeoTrellis M4 (pictured below) devices from Adafruit:

https://learn.adafruit.com/adafruit-neotrellis/
https://learn.adafruit.com/adafruit-neotrellis-m4/

[image]

These devices are usable as relatively inexpensive ($60), small, self-contained (low fi) music synthesizers or USB MIDI controllers with a grid of 4x4 or 4x8 RGB LED buttons.

Here is a quick breakdown of the hardware:
- Microchip ATSAMD51 120MHz Cortex M4 ([ATSAMD51J19A](https://www.microchip.com/wwwproducts/en/ATSAMD51J19A) perhaps?)
  - 12-bit DAC/ADC
  - 512 KB flash
  - 192 KB RAM
- 8 MB SPI FLASH chip 
- [Analog Devices ADXL343 triple-axis accelerometer](https://www.analog.com/en/products/adxl343.html)

tarcieri, 2018-12-27:
Tangibly I'm interested in hearing from anyone who has developed Rust applications leveraging any of this hardware, and also collaborating on a Rust crate which presents some high-level interfaces to the device's functionality.

RandomInsano, 2018-12-30:
This sounds like fun. I've got a few random things <https://gist.github.com/RandomInsano/49148e290b5b3d8650b0d600c3e2dd2b> going right now so I can't promise anything but reach out if you're not sure where to start.

I've been pretty far out of the loop lately so I'm not sure where the USB client crate is for that micro controller at the moment, but the rest of the features of the board look very doable.

jonwingfield, 2018-12-30:
Hi Tony,

There is already some active work going on for the ATSAMD21/51 family of micros at https://github.com/atsamd-rs/atsamd. Most of the work currently focuses on the 21, but there has been some started on the 51, and I think it would be a relatively simple thing to get going once you're familiar with the basics (SVDs, HAL, etc). I'd recommend opening a discussion there.

tarcieri, 2019-01-19:
Thanks! I've opened an issue over there about ATSAMD51 support: https://github.com/atsamd-rs/atsamd/issues/38

tarcieri, 2019-01-22:
Here's a PR to add an ATSAMD51 HAL: https://github.com/atsamd-rs/atsamd/pull/31

I've opened a separate issue about DAC support, documenting what AdaFruit's code does to utilize the DAC, as well as the corresponding Rust ATSADM21 code

tarcieri, 2019-03-03:
I've opened a PR to add a board definition here:

https://github.com/atsamd-rs/atsamd/pull/40

...and with that, I think I can close out this issue. Thanks for the help!

closed 2019-03-03 by tarcieri

tarcieri, 2019-03-04:
One last thing... here's Rust blinking all of the LED buttons! 🎉 

[image]

tarcieri, 2019-03-04:
Okay I lied. Last but not least here is the board support crate: https://crates.io/crates/trellis_m4