rust-embedded/wg #288: Core Cortex-R crates

japaric, 2019-01-06:
This issue is for tracking progress on developing `cortex-r-*` crates that will
be maintained by the WG.

The current plan is

- Make the [`cortex-r-rt`] crate more generic and move it into the WG org. Issue
  paoloteti/cortex-r-rt#1 is being used to discuss how to make that crate more
  generic.

[`cortex-r-rt`]: https://github.com/paoloteti/cortex-r-rt

- Move the [`dcc`] and [`panic-dcc`] crates pretty much as they are into the WG
  org. PR #287 needs to be approved by @rust-embedded/cortex-r before we make
  the transfer.

[`dcc`]: https://github.com/japaric/dcc
[`panic-dcc`]: https://github.com/japaric/dcc/tree/master/panic

- Create an empty [`cortex-r`] repository in the org and progressively expand
  its API surface. When we are happy with the API we'll publish it on crates.io.

[`cortex-r`]: https://github.com/rust-embedded/cortex-r

The motivation for starting from scratch is two-fold: we can enforce that the
crate always compiles on stable (with its default feature set) and we can have
@rust-embedded/cortex-r review each piece of API that gets added to it.

cc @rust-embedded/cortex-r @kenkeiter

paoloteti, 2019-01-06:
@eddyp r52 is available under NDA or for approved partners only. So I prefer to wait a bit.
@japaric I’ll back tomorrow at office and I’ll review your proposal. Thanks

japaric, 2019-01-12:
@eddyp v8r is not suppoted in rustc atm. The crates can't support something that rustc doesn't support. I don't think it would be too much work to update the crates to support v8r, when available, though.

eddyp, 2019-01-15:
@japaric I understand. I don't have experience with older r-s, but I have access at work to an r52 implementation, and was thinking about experimenting with rust, that's why I was interested in r52.

japaric, 2019-01-22:
Triage:

> Move the dcc and panic-dcc crates pretty much as they are into the WG
org.

The arm-dcc repository has been moved into the org: https://github.com/rust-embedded/arm-dcc

---

@eddyp if you are up to it you can make a PR to add a v8-r target to rustc. You can use PR rust-lang/rust#56000 as a reference.

paoloteti, 2019-02-13:
I just released [silicon-cr4](https://github.com/paoloteti/silicon-cr4) that is a collection of workarounds for few silicon errata on Cortex-R4/R4F MCUs. I use it on my TMS570 BSP I started to port to beta/stable. So if you all can review and if everything is ok may be We can decide to move the repo under the `rust-embedde/cortex-r` umbrella.