rust-embedded/wg #291: Embedded WG meeting, January 15

japaric, 2019-01-08:
[8-9 PM CET][0] (Berlin time) on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2019-01-15,420,b8jj

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

Agenda.

- Prioritize Rust requests #269 

- @jamesmunns Help with the newsletter

- @jamesmunns Free form discussions?

- @korken89 rust-embedded/cortex-m-rt#139

- [Issues nominated for discussion, if any](https://github.com/rust-lang-nursery/embedded-wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

japaric, 2019-01-08:
> Prioritize Rust requests #269

If you have opinions on this and can't make it to the meeting, feel free to write down your thoughts in #269. We'll include them in the discussion.

jamesmunns, 2019-01-09:
Re: my item "Free Form Discussions", I talked a little bit on IRC [after the meeting](https://mozilla.logbot.info/rust-embedded/20190108#c15805224-c15805313), but in summary:

I'd be interested in having regular meetings that are specifically less "task completion" focused, but more of an "open forum" to "discuss individual successes and stumbling blocks". This would be a time slot to discuss projects we are working on, small papercuts (or big blockers) that are preventing progress on a library or project, or just for people to share things they are particularly interested.

I'd like to have this as a scheduled time, rather than "just talk in IRC", since I believe there is value in getting people's feedback and discussions in real time, in addition to async discussions.

I'd also like to see this as a place for non wg members to have an opportunity to introduce themselves, ask questions, etc. The WG meetings are open, but if they don't have anything to contribute to the agenda items (which are typically for deep library discussion, compiler issues, etc.), they are not able to add much to the discussion (yet!).

I'd say the major questions would be:

1. When do we have this meeting? 
    * Is it replacing a normal meeting, or in addition to our normal meetings?
    * How often should we have this meeting, weekly, monthly?
    * If separate, should it be similar to the current meeting (right before/after?), or very different (to accommodate people in different time zones/schedules?)
    * Rigid timings, e.g. 1 hour "required attendance", or a more freeform "open house", e.g. "try to show up some time in this 3 hour slot to discuss fun stuff!"
2. Should this meeting have any defined structure or moderation?
    * Totally free form?
    * Defined show+tell spots + some free form time?

I'm very open to ideas, the core things I want to drive are:

1. Giving people new to embedded rust a way to meet people, and feel like part of the community
2. Cross pollination of ideas between teams in the wg, which may not interact with each other very often outside of meetings
3. Increase exposure of the problems we are facing, and the cool things we are building in the rust-embedded ecosystem

HarkonenBade, 2019-01-15:
Would it be possible for there to be a bit of discussion on https://github.com/japaric/bare-metal/pull/15 as it seems to be generating a fair bit of back and forth that may benefit from a larger structured discussion, sadly I don't think I will be able to be present, but I would be interested to read the resulting discussion.

japaric, 2019-01-17:
This meeting happened! Minutes and logs can be found in PR #295.

The agenda for the next meeting is at #296.

@HarkonenBade I have added your item to next week agenda.

closed 2019-01-17 by japaric