rust-embedded/wg #300: Embedded WG meeting, January 29

japaric, 2019-01-29:
[8-9 PM CET][0] (Berlin time) on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2019-01-29,420,b8jj

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

Agenda.

- @korken89 rust-embedded/cortex-m-rt#139

- @harkonenbade  ~~japaric/bare-metal#15~~ rust-embedded/wg#294

- [Issues nominated for discussion, if any](https://github.com/rust-lang-nursery/embedded-wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

korken89, 2019-01-29:
Sorry, wrote in the old call - I might not be able to join this meeting, sorry for the inconvenience! If you want, we can move by part of the agenda to the next meeting.

hannobraun, 2019-01-29:
I won't be making it today. Have a productive meeting!