rust-embedded/wg #301: CI review and policy

jamesmunns, 2019-01-31:
We should probably make a pass of all owned repos, and decide on a policy for CI. This is so that we have consistent policies for what our CI builds should do. In general, I would propose the following:

* For on-commit or on-PR builds, make sure the following are enabled:
    * Build against the stable version of Rust
    * Build against the beta version of Rust
    * Build against the nightly version of Rust, but allow failures
* For notifications, each CI repo should be set up to [notify via email on failures](https://docs.travis-ci.com/user/notifications/#configuring-email-notifications)
    * The notifications should go to the team email that owns that repo (e.g. cortex-m@, resources@, ...)
* Where reasonable, we should be running `cargo test`, at least for the doctests.

jamesmunns, 2019-01-31:
We should also document the policies that we agree on in the Ops repo.

adamgreig, 2019-01-31:
We should also consider adding Travis cron jobs to build repos each day or week to help detect bit-rot early, rather than next time someone tries to open a PR.

nastevens, 2019-02-05:
There's been a lot of really good work over at [crate-ci](https://crate-ci.github.io/), including information on setting up Travis CI, Dependabot, and more. Not saying that we would need to follow every suggestion, but this could be a good piece of reference material. It may even be worth contributing back information about cross-compiled/embedded builds to make it a more complete reference!

japaric, 2019-02-05:
https://github.com/rust-embedded/wg/blob/master/ops/post-transfer.md has some notes on .travis.yml and bors setup

japaric, 2019-02-12:
We discussed this last week so I'm going to remove the nominated label. @jamesmunns if you want to discuss this again in today's meeting feel free to bring it up during the meeting.