rust-embedded/wg #302: Embedded WG meeting, February 5th

jamesmunns, 2019-01-31:
[8-9 PM CET][0] (Berlin time) on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2019-02-05,420,b8jj

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

Agenda.

- @jamesmunns [CI Policy Discussion](https://github.com/rust-embedded/wg/issues/301)

- [Issues nominated for discussion, if any](https://github.com/rust-lang-nursery/embedded-wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

jamesmunns, 2019-01-31:
@japaric I went ahead and added this so I could add an agenda item. Please let me know if I missed anything :)

japaric, 2019-02-02:
@jamesmunns LGTM, thanks for filling this!

japaric, 2019-02-10:
This meeting happened! The logs and minutes can be found in #309.

The agenda for the next meeting is in issue #310

closed 2019-02-10 by japaric