rust-embedded/wg #305: Discord channel

japaric, 2019-02-06:
TL;DR do we want to keep the #wg-embedded discord channel up or take it down?

When the Rust org started their experiment with Discord they created a channel
for every (domain) working group. However, most of the embedded WG members are
only / mostly active on IRC and the Discord channel has low activity, whereas
the IRC channel has seen a big increase in activity these past few months.

I talked to a member from the infra team if it would be possible to set up an
IRC <-> Discord bridge to cater to the people that prefer Discord over IRC
(let's *please* not discuss Discord vs IRC here). The answer was no. Thing is
Discord is used / preferred because they have better moderation tools than IRC
(e.g. requires login). If they made a bridge then it would be hard to moderate
the channel because the bridge will be a single user / bot, which mods can't ban
/ kick because it would take down the bridge.

"Can we set up a bridge only for the embedded WG? Moderation would be left up to the WG members"  If we go that way users may request that the bridge is also set up for other teams / WG and it would be hard to say no ("why just the embedded WG?") and then we'll run into the moderation issue mentionod above.

There were also concerns about where it's actually OK (it's certainly possible) to set up a bridge since the the Discord service agreement is not explicit about being against it.

Something else that was mentioned is that we have the choice to take down the embedded
Discord channel if we (the embedded community) are not using it. One of the Rust
teams does this; they do all their meetings / user interactions on IRC.

So the question, given the above information, is whether we want to keep the
#wg-embedded Discord channel up or take it down.

@rust-embedded/all thoughts?

therealprof, 2019-02-06:
I'm using IRC because I that's what people told me to use not by choice. I find IRC somewhat inconvenient nowadays because it's easy to lose/miss history when not running a custom always-on client.

Disasm, 2019-02-06:
@therealprof It's hard to lose history for EWG room because it's [here](https://mozilla.logbot.info/rust-embedded/).

therealprof, 2019-02-06:
@Disasm As in: I can just open my client, browse through the history and reply...

adamgreig, 2019-02-12:
My vote is to take the Discord channel down rather than leaving it online but unused.

korken89, 2019-02-12:
> My vote is to take the Discord channel down rather than leaving it online but unused.

I agree.

therealprof, 2019-02-12:
My vote would be to use Discord, potentially instead of IRC.

thedodd, 2019-02-12:
There is definitely a large group of individuals out there that would love to be more involved, but IRC keeps them at bay. IRC is great if the goal is to limit accessibility and involvement for this community. 

I won’t go into the reasoning, but I would definitely vote to just make discord official.

No flame. Just feel like this is important to folks and think that we should analyze these aspects as well.

thejpster, 2019-02-12:
My problem with Discord is it's a third-party service running a proprietary protocol. We have no service level agreement with them, nor any comeback if they decide to do something we don't like. Yes, I'm aware of the irony that I make this complain in a Github issue tracker but I don't think adding more proprietary services makes things better. And if I fall out with Github, it's easy enough to pick everything up and put it in Gitlab, or some other service. How would we migrate away from Discord if we felt we needed to?

IRC has its problems, but it's an open standard, and we log the channel so people can review history. If someone can point me at another open standards based chat tool, then I'll happily consider it.

kevinmehall, 2019-02-12:
A couple Rust teams are on Zulip at https://rust-lang.zulipchat.com, which is open source.

Disasm, 2019-02-12:
@thejpster XMPP is an open standard, but multi-user-chats are as "convenient" as IRC ones.

cr1901, 2019-02-13:
My bandwidth is spread too thin with IRC and Twitter already. I am not able to multitask chat clients/applications as well as I used to in my early 20s. If people are moving to Discord, I would much prefer a bridge so I can keep using IRC.

mathk, 2019-02-13:
IRC with matrix bridge work pretty much like discord.

therealprof, 2019-02-13:
@mathk Nowadays you can make everything look like everything. 

Discord is an official Rust communication channel just like IRC. One should not have to resort to a (likely paid) third-party service just to make communication possible (also I'm feeling somewhat uneasy having a proprietary third-party service know my operator credentials); native IRC has a lot of drawbacks including not providing history and being a total clusterfork on mobile/multiple devices. I've happily used IRC for over 20 years now but it's time to move on to something more modern and convenient.

andre-richter, 2019-02-13:
We are having strong opinions on both sides, it doesn't look like we will arrive at a situation that makes everybody happy soon.

I have to chime in on @therealprof 's point about mobile and IRC. I haven't found anything useful yet that is free and not a maintenance horror. Stuff like IRCCloud only works properly if you pay for it.

But we are in the age of mobile. While it might not be suitable to have long discussions over mobile regarding programming topics, at least I would want to have push messages on my mobile notifying me that my attention might be needed. That's just how it works nowadays. Then I can still decide to go to the next machine with a keyboard and answer if I'm in the mood or not.

Yatekii, 2019-02-16:
> I haven't found anything useful yet that is free and not a maintenance horror. Stuff like IRCCloud only works properly if you pay for it.

Quassel. Works absolutely great!

> IRC is great if the goal is to limit accessibility and involvement for this community.

This is 100% false. IRC has a very low entry barrier. I would install and use an OSS chat client without any hesitation. I surely wont for something like discord.

japaric, 2019-02-17:
> let's _please_ not discuss Discord vs IRC here

_Moderation note_: I'd like to remind you about this line I included in the issue description. This issue is about deciding what to do with the existing discord channel given that there's nearly zero presence of EWG members over there. This thread is *not* about A vs B; please take those discussions elsewhere.

---

Regardless of anyone's preferences, I would say that a hard requirement to keep the #**wg**-embedded channel open (with that name) is that there *must* be EWG members actively monitoring and participating in that channel.

If there are not enough members that can to do that (because e.g. they are already busy monitoring GH, IRC, the user forums and/or twitter) then another option might be (\*) to rename the #wg-embedded channel to #embedded and make it a place where people can ask questions about embedded Rust, *but* noting that IRC is where the EWG operates and has its meetings.

(\*) provided that Rust core / infra team approves the idea

EWG members: I'd like to know if any of you has the bandwidth to actively monitor a discord channel.

People that would rather use discord than IRC: I'd like to know if an #embedded channel where the EWG may not be present / active would be useful to you.

therealprof, 2019-02-17:
@japaric I'm actively monitoring the discord channel.

Disasm, 2019-02-17:
@japaric IMO if we do not have enough resources for managing the #wg-embedded Discord channel and we have a hard requirement about this, we should shutdown the channel. If anybody wants to improve accessibility and create an EWG Discord channel, they can do so. I do not see any pros for using that official original channel name. We can create a channel with any other name and show it in the [Contact section](https://github.com/rust-embedded/wg#contact).

therealprof, 2020-08-11:
The discord channel is low profile and a few of us are chiming in there every now and then. I think we can declare this a non-issue and close this topic.

closed 2020-08-11 by therealprof