rust-embedded/wg #310: Embedded WG meeting, February 12

japaric, 2019-02-10:
[8-9 PM CET][0] (Berlin time) on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2019-02-12,420,b8jj

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

Agenda.

- @japaric Updates from Rust All Hands

- [Issues nominated for discussion, if any](https://github.com/rust-lang-nursery/embedded-wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

hannobraun, 2019-02-11:
Most likely, I won't be able to make it to the meeting tomorrow.

japaric, 2019-02-18:
This meeting happened! The minutes and logs can be found in PR #314.

The agenda for the next meeting can be found in #318

closed 2019-02-18 by japaric