rust-embedded/wg #311: Easy firmware flashing

Disasm, 2019-02-15:
We need some tool to make firmware upload process as easy as running "cargo flash".
Possible functions:
* Automatic detection of programmer hardware
* Building additional crates for interfacing programmer hardware
* Download additional toolchains/openocd
* Configuration (with Cargo.toml or separate file)
* Building the user's project before flashing it

Original comments:
https://github.com/rust-embedded/wg/issues/256#issuecomment-445916750
https://github.com/rust-embedded/book/issues/12#issuecomment-411958975

Yatekii, 2019-03-12:
@therealprof and I have started working on this some time ago: https://github.com/Yatekii/probe-rs
Atm it's very ARM specific and not all that well finished & polished, but we are making progress.
The idea would be to have a foundation lib to make it easy to write tools like `cargo flash`.

Disasm, 2019-03-13:
@Yatekii sure, it will be helpful. However, it seems like we still need a gdb layer for debugging. Do you have any ideas about using gdb with `probe-rs`?

therealprof, 2019-03-13:
@Disasm If someone wants to write a gdbserver protocol adapter, that'd be one option. Better ideas (IMHO, of course would be):
- Provide a simple command line tool (something I'm working on)
- Have a GDB/MI protocol adapter
- Have a VSCode plugin
- Nothing of the above and instead have develop a kickass native Rust debugging application with direct compiler support

ryankurte, 2019-03-19:
Hey totally agreed it'd be excellent to make this easy! A couple of thoughts:
- VSCode + [Cortex Debug](https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug) is interesting on the ARM side. Loads SVDs and shows memory nicely, talks to a few different types of debuggers, it might be worth considering how that could be extended for other targets?
- It's probably worth looking at [bobbin-cli](https://github.com/bobbin-rs/bobbin-cli) which can do some of this already

Yatekii, 2019-03-20:
Definitely cool projects you have here!

The problem at least for myself is, that I'd like to eliminate unnecessary layers like GDB, which all of those tools don't. They basically encapsulate the bazzillion tools there is. Each and every of them has it's own quirks and implementations of the GDB protocol. Flashing might or might not work, etc.

I will definitely have a look at those projects, especially the Cortex Debug one, but on the long run, I would love to have a standalone tool without the layer cake.

ejpcmac, 2019-03-20:
@ryankurte Can you flash with Cortex Debug? I am currently using a pre-launch task to do this in C/C++, it would be neat to avoid this.

ryankurte, 2019-03-22:
@Yatekii interesting! i can see the benefit of not having the layers and quirks (though personally might be a little attached to GDB).

@ejpcmac i've been doing the same. there is however an option to pre-run gdb commands on connect in the launch configuration, in which case something like `["halt", "load"]` might do it.

eldruin, 2020-10-06:
[probe-rs](https://probe.rs) has made tremendous progress and provides `cargo flash` and `cargo embed` which work very well. Further ideas can be sent there.

closed 2020-10-06 by eldruin