rust-embedded/wg #315: Not yet awesome embedded Rust

japaric, 2019-02-17:
This issue is to coordinate the creation of a "not yet awesome embedded Rust"
list, which would be a list of things missing in the ecosystem -- things like
crates, tools and documentation.

This list is meant to serve as a bulletin board for people looking what to work
on. Where applicable, each item in this list would also contain a list of WIP
crates, examples, etc. so people can check them out and give feedback to the
authors.

Note that the EWG will not necessarily work on the items in this list; it will
only maintain the list.

---

Some thoughts:

I think we should keep the structure of this list to a minimum: for example we
could start with a flat list and later on group similar items in categories.

Anyone should be able to submit an item to the list but, IMO, we should review
that the request is reasonably scoped / sized and actionable by the community.
Some *counter* examples of these properties:

- An RTOS written in Rust

The scope is too broad: the design space of RTOSes is large with many
trade-offs, and different people expect different things from an RTOS (e.g. a
network stack). More narrowly scoped requests would be "bindings to
$SOME_PARTICULAR_RTOS" and "a thread-based concurrency framework".

- Certified Rust compiler

This is *not* reasonably sized. This project requires engineer-years levels of
work. Something reasonably sized should be on the order of, or below, a few
developer-months for a first release.

- Support for Infineon TriCore microcontrollers

This is *not* actionable. It is blocked on LLVM support for the architecture and
changes in rustc.

cc @jamesmunns @rust-embedded/resources thoughts?

jamesmunns, 2019-02-19:
I think this makes sense.

I think how the WG can help with this is making these kind of review comments during the PR process. I think this is what you mean by this:

> Anyone should be able to submit an item to the list but, IMO, we should review
that the request is reasonably scoped / sized and actionable by the community.

So more concretely, I think the process should be:

**Adding an item:**

1. Anyone submits a PR, making a request for something that is needed in the ecosystem
2. Someone from the WG (Ecosystem team?) reviews the PR, to make sure that the request is:
    * Request is clearly stated, linking to relevant documentation, such as a whitepaper, protocol definition, datasheet, etc
    * "Success Criteria" is defined, such as "Add PWM support to the iMXRT1050 HAL"
    * Possible using today's Rust (not blocked by LLVM impl, rustc impl, etc.)
    * Broken into reasonable work packages, such as "Create HAL for XYZ chip", not "support all boards from ABC vendor"
3. Once the review checklist is complete, and any fixes are made, the item is added to the "TODO" list

**Marking an item as WIP**

This is meant to be a "low barrier to entry" update to link to anyone working on the items on the TODO list, to help guide people to contribute to existing WIP efforts. The process would be:

1. Anyone submits a PR, adding a link to a WIP impl of a TODO item.
2. Someone from the WG (Ecosystem team?) reviews the PR, to make sure the WIP is:
    * Links to the source project on GitHub/GitLab/etc.
    * States whether outside contributions are welcome (yes/no/unknown)
    * Relevant to the TODO item
    * More than just an empty repo or a placeholder (e.g. just a "hello world" `bin.rs`/`lib.rs`)
3. Once the review checklist is complete, and any fixes are made, the WIP crate is added under the TODO item.

**Removing the item from the TODO list**

1. Anyone submits a PR, removing the TODO and WIP items from the list
2. Someone form the WG (Ecosystem team?) reviews the PR, to make sure the impl:
    * Addresses the original TODO request fully
    * The crate that addresses the TODO has been added to the Awesome Embedded Rust list
    * The crate is published on crates.io
3. Once the review checklist is complete, and any fixes are made, the TODO item and all WIP crates are removed from the list. :confetti_ball: :tada:

jamesmunns, 2019-02-19:
We may want to link both the "awesome" and "not yet awesome" list on https://docs.rust-embedded.org/.

japaric, 2019-02-27:
@rust-embedded/resources I want to propose adopting the [process](https://github.com/rust-embedded/wg/issues/315#issuecomment-465177817) proposed by @jamesmunns (with "someone from the WG" = resources team) and creating the not yet awesome embedded Rust right after that's approved. Let's vote:

- [ ] @adamgreig 
- [ ] @andre-richter 
- [ ] @jamesmunns 
- [x] @japaric 
- [ ] @korken89 
- [ ] @ryankurte 
- [ ] @thejpster 
- [ ] @therealprof

japaric, 2019-03-07:
This proposal has been approved and the NYAER list now exists at https://github.com/rust-embedded/not-yet-awesome-embedded-rust :tada:

closed 2019-03-07 by japaric

brainstorm, 2023-03-18:
Those look like they are TriCore: https://www.infineon.com/cms/en/about-infineon/press/market-news/2023/INFATV202303-076.html