rust-embedded/wg #316: Embedded Rust "libs blitz"

japaric, 2019-02-17:
This issue is to discuss doing a embedded Rust version of ["The Rust Libs
Blitz"](https://blog.rust-lang.org/2017/05/05/libz-blitz.html).

The idea would be to review library crates: their implementation (\*), their API
surface, their use of unsafe code and their test suite. In the case of early
stage libraries we would work through API design questions with the authors
(e.g. "how do I design my API so that the user cannot call method X before doing
Y").

(\*) the parts of the implementation that don't require the reviewers being
familiar with specialized knowledge like the USB specification.

## Strawman proposal

- Create a repository for this

- Request for reviews are to be submitted as issues

- The team in charge of the repository reviews the requests and picks one crate
  to work on for the next two weeks.

- In the case of established crates:
  - During the first week the team produces a document that contains a list of
    suggestions.
  - The next week the team discusses their suggestions with the crate authors.

- In the case of early stage crates
  - During the first week, the team and the crate authors work through a high
    level specification of the crate, previously produced by the crate authors,
    to develop potential APIs and usage examples.
  - The next week is spent discussing the trade offs of the different APIs and
    giving the authors suggestions on implementation strategies and testing.

- Once the review is done, the team takes a 2-week "break" during which they
  decide which crate next to work on. This break period also serves as a
  "buffer": the team may, for example, extend the review process of a crate to 3
  weeks (and only rest 1 week) to account for members being unavailable mid way
  the review process.

- Crates requested in the 2019 wishlist survey will be given higher priority
  when deciding what crate to review next.

- As the team gains experience they'll produce guidelines for writing libraries
  covering design patterns (e.g. singletons) and best practices (e.g. testing).

## Unresolved questions

- We need to pick a different phrase / term for this. "blitz" has negative
  connotations.

jamesmunns, 2019-02-19:
Bikeshed, maybe "build out"?

jamesmunns, 2024-06-11:
Closing for inactivity as part of the 2024 triage effort.

closed 2024-06-11 by jamesmunns