rust-embedded/wg #318: Embedded WG meeting, February 19

japaric, 2019-02-18:
[8-9 PM CET][0] (Berlin time) on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2019-02-19,420,b8jj

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

Agenda.

- Reminders / announcements (10 mins)

- Triage team (10 mins) - https://github.com/rust-embedded/wg/pull/144

- Showcase (10 mins) - https://github.com/rust-embedded/showcase/pull/6

- Biweekly newsletter (20 mins)

- [Issues nominated for discussion, if any](https://github.com/rust-lang-nursery/embedded-wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

flip111, 2019-02-18:
Please change the title in google calendar from "Weekly meetings" to "Rust Embedded meeting".

vertexclique, 2019-02-22:
One of the topics for discussion might be creating a RTOS team.

mathk, 2019-02-22:
An other topic would be to discuss on how to avoid duplication in the different stm32xxxxx-hal. There is some hal that are more frequently updated like the stm32f4xx-hal and some lagging behind ,  I see a lot of duplicate effort going on on the different hal.

mathk, 2019-02-22:
An other topic that I am interested to solve is how can we provide clean abstraction to enable different power mode and be able to switch from one mode to an other and keeping peripheral safe when entering and leaving mode.

Disasm, 2019-02-24:
This meeting happened! The minutes and logs can be found in PR #322.

The agenda for the next meeting can be found in #321

closed 2019-02-24 by Disasm