rust-embedded/wg #321: Embedded WG meeting, February 26

Disasm, 2019-02-23:
[8-9 PM CET][0] (Berlin time) on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2019-02-05,420,b8jj

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

Agenda.

-  Reminders / announcements

- @vertexclique RTOS team - https://github.com/rust-embedded/wg/issues/318#issuecomment-466368407

- @mathk *-hal deduplication, support for different power modes

- @Disasm Proper handling of write-only SPI channels (https://github.com/rust-embedded/embedded-hal/pull/121), semantics of SPI `read()` function (https://github.com/rust-embedded/embedded-hal/pull/120)

- [Issues nominated for discussion, if any](https://github.com/rust-lang-nursery/embedded-wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

flip111, 2019-02-24:
@Disasm can you add this too ?  https://github.com/rust-embedded/wg/issues/22#issuecomment-464929863

japaric, 2019-02-27:
@vertexclique

> RTOS team

We discussed this during the last meeting and the consensus was that we do *not*
want to create an RTOS team right now. The reason is that we want to avoid
maintaining crates tied to any particular vendor in the org because that could
snowball into us maintaining crates for all the RTOSes out there -- this is also
the reason why we don't maintain any HAL implementation in the org.

We are open to maintaining things like traits that abstract over different
RTOSes but before those exist different RTOS bindings / implementations must
exist first. Thus we think that developing and maintaining RTOSes / RTOS
bindings should be left up to the community. However, we'd be glad to raise
awareness about on-going community efforts in our newsletters, (not-yet) awesome
lists, etc.

flip111, 2019-02-27:
@japaric perhaps it's possible to create some hardware agnostic parts of the RTOS. Maybe a scheduler? Or just having an inventory of which parts would be needed would already be useful. Maybe some people would like to think about this. Not as a team capacity but just investigation.

Disasm, 2019-02-28:
This meeting happened! The minutes and logs can be found in PR #325.

The agenda for the next meeting can be found in #326

closed 2019-02-28 by Disasm