rust-embedded/wg #326: Embedded WG meeting, March 5

Disasm, 2019-02-28:
[8-9 PM CET][0] (Berlin time) on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2019-03-05,420,b8jj

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

Agenda.

-  Reminders / announcements

- @flip111 svd2rust-like tool for external device specification - https://github.com/rust-embedded/wg/issues/22#issuecomment-464929863

- [Issues nominated for discussion, if any](https://github.com/rust-embedded/wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

flip111, 2019-03-02:
Maybe it would be nice to have bindings for https://github.com/littlevgl/lvgl micropython already has bindings for it

japaric, 2019-03-12:
This meeting happened! The logs and minutes can be found in #332.

The agenda for the next meeting is at #331

closed 2019-03-12 by japaric