rust-embedded/wg #335: How can Vec<T> be using on STM32?

johnthagen, 2019-03-19:
Cross post from https://github.com/rust-lang/rfcs/pull/2480#issuecomment-473339244

When evaluating using Rust for an embedded project at work, not having dynamic containers was a big drawback at the time compared with C++11/14 where the `gcc-arm-embedded` toolchain had versions of `std::string` and `std::vector<T>`. (C++ has it's own set of drawbacks, like exception-based error handling APIs, but in this one, fundamental case, it had a leg up).

Is there a way (even it's nightly) to use `Vec<T>` on something like the [STM32F3DISCOVERY](https://rust-embedded.github.io/discovery/)? It was a hard sell to recommend Rust over C++ without being able to safely use the heap.

therealprof, 2019-03-19:
https://crates.io/crates/heapless

johnthagen, 2019-03-19:
@therealprof Thanks, yeah I had seen `heapless` and I changed the title to be more clear that we'd like to actually use the heap for true dynamically sized containers. `heapless` requires a fixed size up front.

therealprof, 2019-03-19:
You can use alloc-cortex-m to create a heap and combine it with heapless.

japaric, 2019-03-19:
@johnthagen what @therealprof said. The quickstart template has an example: https://github.com/rust-embedded/cortex-m-quickstart/blob/master/examples/allocator.rs

japaric, 2019-03-19:
(you don't need to specifically use alloc-cortex-m; any global_allocator will do)

johnthagen, 2019-03-19:
@japaric Is this mentioned in the Discovery Book? And if not, would you accept a PR to add it? I think this would be a really great addition!

johnthagen, 2019-03-19:
(For anyone else finding this issue in the future, [the embedded book has some discussion on this as well](https://rust-embedded.github.io/book/collections/index.html).)

therealprof, 2019-03-19:
I think this would be more appropriate for the book and/or the embeddonomicon with a reference in the discovery book. Any contributions in that directions would certainly be welcome.

jamesmunns, 2019-03-19:
I think we should also note (last time I checked) that `alloc-cortex-m` requires nightly rust to operate.

johnthagen, 2019-03-19:
> I think we should also note (last time I checked) that alloc-cortex-m requires nightly rust to operate.

Correct, this RFC is currently being discussed to potentially stabilize `alloc`: https://github.com/rust-lang/rfcs/pull/2480

johnthagen, 2019-04-22:
`alloc` is on track to stabilize for Rust 1.36.0: https://github.com/rust-lang/rust/pull/59675

MarcoIeni, 2020-06-14:
So:
- The answer to the question is to use the `alloc` crate.
- [This](https://github.com/rust-embedded/discovery/issues/182) issue was opened in order to mention this fact in the discovery book.

I think that no action is required here anymore and this issue could be closed.

jonas-schievink, 2020-06-14:
Yup, closing then. Thanks!

closed 2020-06-14 by jonas-schievink