rust-embedded/wg #346: Oxidize Coordination

jamesmunns, 2019-04-16:
A number of /wg members will be attending [OxidizeConf](https://oxidizeconf.com) in Berlin. If you plan to be here, let's try and coordinate time for the teams to meet and discuss things, such as #265.

CC @rust-embedded/all

jamesmunns, 2019-04-16:
I will be there for the conference and both Impl days (Sunday and Monday), though I may be running around organizing things.

korken89, 2019-04-16:
I'll be there until Sunday afternoon.

therealprof, 2019-04-16:
I'll be there on Saturday only.

thejpster, 2019-04-16:
I'm there all four days

Disasm, 2019-04-16:
I'll be there for all four days.

japaric, 2019-04-16:
I'll be there for the four days (but I'm not attending any workshop).

hannobraun, 2019-04-18:
I'll be there all 4 days. I'll also have some time the day before (late afternoon and evening) and the day after (morning and possibly early afternoon), in case anyone wants to meet.

adamgreig, 2019-04-21:
I'm there from Friday evening until Monday afternoon.

adamgreig, 2019-04-28:
This happened :tada:

closed 2019-04-28 by adamgreig