rust-embedded/wg #347: Embedded WG Meeting, April 23

jamesmunns, 2019-04-16:
[8-9 PM CET][0] (Berlin time) on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2019-04-02,420,b8jj

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

Agenda.

-  Reminders / announcements
- [Issues nominated for discussion, if any](https://github.com/rust-embedded/wg/labels/nominated)
- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

Dylan-DPC, 2019-04-17:
Have a few points about cross: 

* New release: The last release of cross was in November 2017 which means users have to depend on latest master branch (which may break on certain architectures) for newer features.

* Qemu failures: Currently build is failing on a few architectures because of QEMU and we haven't found a way to solve the issue and it is blocking other unrelated PRs (we could disable the architectures but that shouldn't be the first option we take) 

I might not be available for the meeting but if you need more clarity on any of the above issues you can ping me on discord

stappersg, 2020-02-27:
Please close this issue  (or say why it is still open).

See also #362 comment from 2020-02-27 which has
> Open issues do drain human energy

closed 2020-02-27 by adamgreig

Dylan-DPC, 2020-02-27:
Sure, especially if the open issues are closed without receiving any update, which means back to square one 😅

Thanks 😃