rust-embedded/wg #348: Network abstractions crate

benjaminjt, 2019-05-01:
Would there be some value in a crate for network layer abstractions (e.g. Ethernet, IP, TCP, etc.) in the same vein as https://github.com/rust-embedded/embedded-hal?

This may a naive view, but it seems to me that it would be very useful for building up an embedded networking stack. For example:

A common `Ethernet` trait implemented by board support and driver crates (such as https://crates.io/crates/enc28j60) would allow TCP crates (such as https://crates.io/crates/enc28j60) to run on arbitrary hardware.

A common `TcpSocket` trait implemented by TCP crates (and maybe even std?) would allow higher level abstractions (such as TLS or HTTP) to be built more easily.

ryankurte, 2019-05-03:
hey interesting idea! seems possible, though i suspect we'd need some more examples to extract the commonality and create something generally useful. my suggestion would be that it might be worth developing the networking bits outside the wg with the goal of adoption once it's demonstrated / useful?

also i think an ethernet peripheral trait (if it can be generalised) is definitely in scope for embedded-hal ^_^

michalfita, 2019-07-10:
Don't forget you need Ethernet at two levels of abstraction, original IEEE 802.3 which can be directly under IP protocol and works today on copper or fibre cable and higher one for everything that mimics Ethernet frames on top of medium like IEEE 802.11 or FrameRelay. The former fits into `embedded-hal` as would address implementation over physical peripheral of the MCU, but then the latter shall be part of network abstractions. Maybe the one from `embedded-hal` could directly act as implementation for the one from network abstraction crate (if that would be the case)?

newAM, 2020-06-07:
Would there be interest in _just_ embedded networking types to start with?  It would not solve the problem; but it would improve portability across existing embedded networking libraries.

Specifically I am thinking there should be types for ipv4/ipv6 addresses and EUI-48/EUI-64 hardware addresses.  These have very well defined standards, and in the case of IP address there is a lot to leverage from [ip.rs](https://doc.rust-lang.org/src/std/net/ip.rs.html) in [std::net](https://doc.rust-lang.org/std/net/index.html).

jonas-schievink, 2020-06-07:
Related: https://github.com/rust-lang/rfcs/pull/2832

ryankurte, 2020-06-07:
ahh glad to see some progress on the move of types to core! supporting both `std` and `no_std` at this time is a right pita without em.

ryan-summers, 2020-06-08:
This is another crate that I've seen used in a few embedded crates: https://github.com/thejpster/embedded-nal

eldruin, 2020-10-06:
We agreed in today's WG meeting that [`embedded-nal`](https://github.com/rust-embedded-community/embedded-nal) is the place for these abstractions.
Further ideas and improvements are welcome over there!

closed 2020-10-06 by eldruin