rust-embedded/wg #353: RNG support in libcore

tarcieri, 2019-06-04:
Of interest to this WG are some developments on [rust-random](https://github.com/rust-random), namely the new [getrandom](https://github.com/rust-random/getrandom) crate and plans to integrate it into `std`:

https://github.com/rust-random/getrandom/issues/21

There's an open issue about `no_std` support for `getrandom` in general, including various ways it could potentially be integrated into `core` instead of `std`:

https://github.com/rust-random/getrandom/issues/4

One approach might be the [addition of a new lang item](https://github.com/rust-random/rand/pull/579#issuecomment-424443465).

Curious of anyone has thoughts on this for embedded device use cases. Since `embedded-hal` generally eschews global state for things like device peripherals, I'm wondering if the existing `rand-core` API may actually be a better fit.

michalfita, 2019-07-10:
One example from my backyard is entropy for encryption (TLS). While `getrandom`'s source is OS, I wonder how reaching TRNG peripheral of some of MCU could work in such case? If I correctly understand proposal from rust-random/getrandom#4 it would require own implementation to let the further code use a standard API. That's definitely worth consideration.

therealprof, 2019-07-10:
One thing that works is instantiate some seedable RNG (like `ChaChaRng`) and either seed it with TRNG data when booting up or periodically feed it with some entry e.g. using an interrupt.

michalfita, 2019-07-11:
@therealprof TRNG in ATSAMS70Q21 is fast enough to use directly as source of random numbers, we drop the need for cycles to calculate any PRNG.

therealprof, 2019-07-11:
@michalfita So what prevents you from using that directly?

michalfita, 2019-07-11:
@therealprof My point was about finding good pattern how to use the peripheral by the abstraction layer, so the use of standardised functions would still work. I'm mostly thinking about portability of the code.

ryankurte, 2020-07-13:
i ran into this recently and ended up with a bunch of mutex and `rng_core` faff to support global use of `rand_core::Rng` objects while supporting platform initialisation. Unsure how to manage periodic re-seeding atm but it's here if it's useful [ryankurte/rust-rand-facade](https://github.com/ryankurte/rust-rand-facade).

i also completely missed that `getrandom` has a feature for defining a global random function [getrandom#109](https://github.com/rust-random/getrandom/pull/109), which _may_ have done the job, but isn't obvious from any documentation, so leaving that here in case anyone else runs into it.

newpavlov, 2020-07-13:
@ryankurte 
Note that this `getrandom` feature is part of the (currently unreleased) v0.2 and is not present in v0.1.

jannic, 2024-06-11:
https://github.com/rust-lang/rust/issues/62079#issuecomment-2148180934 suggests closing the ticket for std.

For no_std applications, https://crates.io/crates/rand_core provides good support, and implementations can be provided by HALs.

So I think this ticket can be closed.

jamesmunns, 2024-06-11:
Agreeing with @jannic, and closing as part of the 2024 triage efforts.

closed 2024-06-11 by jamesmunns