rust-embedded/wg #356: Embedded WG Meeting, Jun 11

therealprof, 2019-06-04:
[8-9 PM CET][0] (Berlin time) on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2019-03-19,420,b8jj

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

Agenda.

- Reminders / announcements
- [Issues nominated for discussion, if any](https://github.com/rust-embedded/wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

therealprof, 2019-06-05:
@jamesmunns NB: I won't be available for this meeting.

jamesmunns, 2019-06-17:
This meeting happened!

closed 2019-06-17 by jamesmunns