rust-embedded/wg #357: [VOTE] Choose the next chat platform

jamesmunns, 2019-06-05:
This is the implementation of [RFC0351](https://github.com/rust-embedded/wg/blob/master/rfcs/0351-chat-vote.md) - voting on a new chat platform. The highest voted platform based on this vote will become the official room used for the Rust Embedded Working Group.

Please read the text of the RFC for the full content, but a quick summary of the rules:

* Voting will begin immediately. Voting will be open for (a little more than) two weeks, concluding at 23:59 CEST on Sunday, 2019-06-23.
* Only selections proposed by the Embedded WG will be considered.
* Votes are made by using the :+1: emote on the comments below.
* Users may vote for multiple options.
* If the top two choices are within 3 votes of each other, there will be a follow on ballot with the two highest ranking options. In case of a real tie in the second vote, the EWG Core team will break the tie.

Please feel free to discuss the options below, **HOWEVER** please keep in mind our [Code of Conduct](https://github.com/rust-embedded/wg/blob/master/CODE_OF_CONDUCT.md), and please refrain from suggesting additional platforms.

jamesmunns, 2019-06-05:
# Option 1: Discord

We would use the [`wg-embedded`](https://discordapp.com/channels/442252698964721669/443151017681485835) room of the Official [Rust Discord Server](https://discordapp.com/channels/442252698964721669/).

jamesmunns, 2019-06-05:
# Option 2: Gitter

We would use the [gitter.im](https://gitter.im/) server based on the `rust-embedded/wg` repository.

jamesmunns, 2019-06-05:
# Option 3: IRC (Freenode)

We woud use the `#rust-embedded` channel on [Freenode IRC](https://freenode.net/).

jamesmunns, 2019-06-05:
# Option 4: Matrix

We would set up a channel on a hosted [Matrix](https://matrix.org) instance.

jamesmunns, 2019-06-05:
# Option 5: Zulip

We would set up a room on a hosted [Zulip](https://zulipchat.com/) instance.

jamesmunns, 2019-06-05:
# END OF OFFICIAL VOTES

Only items listed above this line will be considered for the official vote.

jamesmunns, 2019-06-05:
# Beginning of Comments

Discussion of the above items begins below here. Information discussed below may be included in the posts above on a periodic basis.

jamesmunns, 2019-06-05:
CC @rust-embedded/all

berkus, 2019-06-05:
### Option 6: Telegram

* Simple
* Lightweight
* Persistent history
* File and other media sharing and storage directly without 3rd party services
* Endless chat branching (offtopic/general/newbies are easily created and cross referenced)
* Searchable history, #tags
* Convenient mobile clients
* Web client exists and is quite usable

mvirkkunen, 2019-06-05:
Whatever is chosen I truly wish it won't be a closed proprietary platform.

Closed means being closed to using your own preferred client software that you can choose and configure to *your* liking above all others, whether that means convenience or accessibility.

Proprietary means not being ran by a community of enthusiasts but rather a faceless corporation, at which point your data and continued existence are at the whim of the company's business decisions and likely a less-than-ideal legal system.

nbraud, 2019-06-05:
@berkus I think you missed James' [end-of-list comment] and an important rule point:
> Only selections proposed by the Embedded WG will be considered.

[end-of-list comment]: https://github.com/rust-embedded/wg/issues/357#issuecomment-498925900

therealprof, 2019-06-05:
@mvirkkunen If it has a web interface it's not closed, you can select whichever browser you want.

IGBC, 2019-06-05:
Discord is a really problematic platform, they have history selling user's data, and a rather arbitrary usage policy that does not really protect Mozilla's community guidelines. Additionally their Linux client is poor reducing accessibility for a lot of users. 

Please Consider this when casting your vote.

therealprof, 2019-06-05:
@IGBC Is there any proof of this? Also which user data are they supposed to sell? Worst case they have your email address and I can tell you they haven't sold mine yet since every service I use gets a unique email address here. Also I'm not sure why people would even consider using a special client (which is nothing but a web browser running a specific application) instead of a regular (or even dedicated) browser.

IGBC, 2019-06-05:
@therealprof  "When you interact with us through the Services, we receive and store certain information such as an IP address, device ID, and your activities within the Services. We may store such information or such information may be included in databases owned and maintained by affiliates, agents or service providers" - Lifted directly from the discord [https://discordapp.com/privacy](Privacy Policy)

As per native clients - I don't mean to be dismissive but I personally don't consider a web browser (or a web browser packaged in an app) to be a good chat experience. Jokes about Slack aside I literally do not have enough ram or battery life to keep every chat platform's webapp in a browser tab and still do my work. `</rant>`

therealprof, 2019-06-05:
@IGBC Those are pretty standard terms and not proof of anything. Worse yet: With IRC **anyone** can gather the same kind of information, not just the operator.

> As for native clients - I don't mean to be dismissive but I personally don't consider a web browser (or a web browser packaged in an app) to be a good chat experience.

I certainly prefer a web browser over a special protocol client, especially with IRC clients which are going the way of the dodo (read: unmaintained bitrot). Ever tried to use IRC on a tablet or phone, Discord in a webbrowser is a godsend in comparison and actually works...

mvirkkunen, 2019-06-05:
@therealprof If a web browser is your preference there's always web IRC clients and IRCCloud.

therealprof, 2019-06-05:
@mvirkkunen Yeah, doesn't address any of the drawbacks of IRC though.

knyzorg, 2019-06-05:
A major benefit of matrix is that we could setup bots to sync matrix activities with discord, irc, telegram, etc fairly easily. 

I for one would really like to be able to use telegram without forcing anyone to get on it as well.

andre-richter, 2019-06-05:
Copy/pasting a point I made in another thread: What annoys me about IRC is that the mobile experience is the probably the worst of all options.

But we are in the age of mobile. While it might not be suitable to have long discussions over mobile regarding programming topics, at least I would want to have push messages on my mobile notifying me that my attention might be needed. That's just how it works nowadays. Then I can still decide to go to the next machine with a keyboard and answer if I'm in the mood or not.

People suggested IRCCloud and Quassel. For the first one I have to pay 40€ (!!!) per year to stay connected. Not an option. Quassel, on iOS at least, also has a payment barrier.

The other options would at least work out of the box on modern devices. That is convenient.

I also do not really see the point of "selling user data". Whatever platform, any bot could just harvest whats being talked and sell it. Its openly accessible after all.

paulvt, 2019-06-05:
In the case of web browsers, I think one can no longer speak of having a non-special protocol client. The free and open web browsers that we all have and love to use nowadays just provide a platform to run those huge, often closed, proprietary special protocol clients. Just... on another VM.

Open platforms with decent open protocols (I'm not really defending IRC here ;)) provide ways to connect with a variety of clients and provide choice.  They might be rough around the edges sometimes, but at least if people flock towards them they will improve. If we put those options aside, and go with some fixed client or app that might not go in the direction you/we want, say in a year or two from now, either we go through this process again or we stick with what we have, often feeling entitled they should make things better, while they actually shouldn't; companies can do whatever they want.
This might seem like a hypothetical case, but there are all too many examples.

Shizmob, 2019-06-05:
> I also feel that people are overrating "openness" quite a bit. Something as "open" as IRC means you cannot have any privacy: without additional (tricky) measures anyone can easily deanonymise anyone, track everyone, impersonate anyone and simply spy on the whole traffic, or even replicate the service and put the replica into a walled garden (or great firewall 😏). Especially with a community which embraces diversity and people from all around globe (even those living in countries oppressing the people based on differing opinions, ethnics, religions, social standing...), something as open as IRC can be a huge problem!

This is a very weird position to take from my perspective. Whether the specifications of a protocol are open for implementation has very little relevance to whether it can be spied upon, which is a content feature of the protocol itself. You can definitely not spy upon all IRC traffic as a random user, and being able to set up your own servers, possibly proxying themselves, is a distinct advantage for people under oppressive regimes where mainstream networks may be blocked. To me, it seems like 'open' relating to content ('everyone can see all traffic') and 'open' relating to the specification ('everyone could implement it') are being confused here.

>Whatever platform, any bot could just harvest whats being talked and sell it. Its openly accessible after all.

This does not apply to people's private conversations typically, or (private) email addresses, which is a valid concern to be had in my opinion. Once people are on a platform, they will inevitably expand some of their activities, including private conversations with other users, to that platform. Being careful with choosing what platform users are provided with is therefor important.

-- 

Regardless, the disadvantages of IRC are distinct and real: the mobile experience is suboptimal, there is no integrated message history and only tangentially permanent identification, and moderation tools are lacking.  This is why I voted for Matrix and would like to propose to everyone to seriously consider it, as it provides for all of the disadvantages of IRC:
* Full message history;
* Permanent identification;
* Good moderation tools, with [a guide in active development](https://matrix.org/docs/guides/moderation);
* Responsive developers, open to feedback; 
* A serviced mobile story with [Riot](https://about.riot.im/);
* An open-to-implement platform not ran by a for-profit corporation with possibly perpendicular interests to the community.

I think it fits in well with the community model, and can be trivially bridged to IRC for people who do prefer that.

joepie91, 2019-06-05:
I just want to emphasize here that the Riot client for Android is *really quite good*. It's quite unlike the typical open-source mobile clients that one might be used to.

taciturasa, 2019-06-06:
Matrix has official IRC bridge functionality, is a FOSS protocol, and could easily be used as a default with IRC as a legacy alternative. It also supports avatars, read badges, embeds and files, voice/video, and persistant history. It it federated as well, meaning your user doesn't even have to be on the same instance to communicate. There's reaaaally no reason to NOT choose it and go with IRC other than to just be curmodgeony.

mvirkkunen, 2019-06-06:
I've upvoted both IRC and Matrix, because a Matrix room with IRC integration would probably provide best of both worlds and satisfy most people. Matrix has all the bells and whistles of a "cloud" service for the mobile first people, and it has great IRC integration for us old people.

therealprof, 2019-06-06:
@mvirkkunen This is not how this works. If the vote is going to IRC you can use Matrix yourself to bridge (and get the worst of both worlds), if it's going to Matrix there won't be any IRC anymore. Voting for IRC if you really want Matrix is likely going to defeat the purpose.

f0x52, 2019-06-06:
@therealprof Why not? Bridges are first-class citizens in Matrix, especially the IRC bridge. You can have Matrix as your main platform, but still allow bridging to an irc channel for everyone that wants to use an irc client

therealprof, 2019-06-06:
@f0x52 This vote is supposed to select **one** protocol to use in the future which will be the official one. There's of course a possibility that some volunteer might set up a bridge to some other system/protocol but this is out of scope for this vote.

Of course people are free to vote however they like, but each vote has equal weight, so if one has a clear preference of Matrix over IRC it does not make sense to vote for both, since this doesn't reflect the preference in the end result, the second place will not receive any consideration by the WG as part of this RFC. Instead if someone is interested in such a solution it would be up to him to propose a RFC implementing such a bridge.

mvirkkunen, 2019-06-06:
@therealprof I understand "Matrix with IRC bridge" isn't a valid option to vote but since both are on the table why dismiss using both? I'm a part of another "techie" community where Matrix and IRC are used equally with great bridging you barely even notice (users show up with their real nicknames, not as some bridge username) and that is there for everyone (no need to set up anything yourself - just join using either protocol). The reason for having both was what I mentioned in my previous comment, and it works.

f0x52, 2019-06-06:
Then I urge everyone who voted for IRC to reconsider voting for Matrix, if they find a Matrix <> IRC bridge acceptable.
From my experience they work quite well on both sides, except maybe for the "user x sent a long message, here's a link", which can be disabled by using a bridge with different config options.

mvirkkunen, 2019-06-06:
@therealpeof (wrt later comment) welp, if the process is this rigid I guess us IRC people will have no option but to vote for IRC and IRC only in order to not lose access, even though another option would perhaps be more considerate for the community as whole.

therealprof, 2019-06-06:
@mvirkkunen Well, there is an RFC and it has been accepted. Discussions have happened and people provided their input (not only to the RFC but also to earlier issues and in meetings); obviously no one found the time or spirit to propose this solution. So yes, now that the process is in motion, we're going to stick with it.

It is perfectly okay for IRC people to vote for IRC, that's why the option is on the list.

It's also okay to preemptively make a proposal about this Matrix <> IRC bridge, this can be discussed in the next meeting and potentially an agreement can be reached to also implement it **iff** Matrix gets the most votes. This may also help convince people to give their vote to Matrix.

TeXitoi, 2019-06-06:
Can you connect to the #rust-embedded channel from freenode using matrix out of the box as there is an official freenode bridge operated by Matrix.org?

f0x52, 2019-06-06:
@TeXitoi Yes, or using third party bridges like the one I run for pixie.town (#freenode_#rust-embedded:matrix.org or #freenode_#rust-embedded:pixie.town)
If IRC is chosen, Matrix users will still be able to connect, if allowed by the moderators.
Having Matrix listed as an official platform (optionally next to IRC) would still be beneficial to new users though, due to it's browser client that is easier to use due to media sending etc.

@mvirkkunen I run matrix-appservice-irc as part of the pixie.town services, and would not mind rust-embedded using that, it's long message to link limit is currently set to 10 lines.

therealprof, 2019-06-06:
@f0x52 

> If IRC is chosen, Matrix users will still be able to connect, if allowed by the moderators.

Today it is also possible to use Matrix to connect irc.mozilla.org. This has all the disadvantages of IRC plus some more quirks of Matrix/RIOT, too so it's not really a good solution IMHO.

TeXitoi, 2019-06-06:
@therealprof can you explain that a bit more? If the official channel is on matrix, is it better to have the bridge the other way?

therealprof, 2019-06-06:
@TeXitoi My understanding of Matrix is that it is a modern and open chat platform (someone will certainly correct me on this) on par with the other (closed) modern alternatives. Means history, privacy, safety, authentication, text correction, emoticons, text markup are all there.

Bridging that to IRC would only throw the people using the bridge 20 years back into into the past while everyone else gets to enjoy the perks of modern communication which sounds quite acceptable to me as long as there's no maintenance burden for the WG.

linuxtim, 2019-06-07:
FWIW...

If it were chosen at some point in the future to bridge an irc channel to a matrix room...  The moderation features of Matrix effectively don't apply to the messages that get bridged into the IRC channel.  This may not be an issue in the real world (I believe it hasn't yet been a problem with the existing IRC channel).

If it did become a problem in the future (to such an extent that the channel mods decide to remove the bridge), then Matrix has a couple of other solutions (e.g. https://github.com/matrix-org/matrix-ircd ) for people who wish to continue to use IRC-only clients to access the matrix room.

linuxtim, 2019-06-07:
For the record (iff matrix is selected, and matrix.org is selected as the server)...  Approx 6 weeks ago, I pre-emptively registered #rust-embedded:matrix.org - which I'm happy to hand over fully to the WG if requested.

f0x52, 2019-06-07:
There's also a weechat plugin that allows you to use matrix directly https://github.com/poljar/weechat-matrix

michalfita, 2019-06-11:
It's interesting Mattermost haven't been taken into consideration.

mmstick, 2019-06-12:
We're using Mattermost for Pop!_OS (1500+ users) and Redox OS (700). Works great.

- https://chat.pop-os.org
- https://chat.redox-os.org (you must request an invite)

taciturasa, 2019-06-12:
The issue with Mattermost is that it only has choice for one client.

mmstick, 2019-06-12:
If that were an issue, Discord wouldn't be a consideration at all. Its TOS expressively forbids third party clients.

Mattermost is fully open source (MIT license), and its API is easy to interact with. Anyone's free to write third party clients for it. [Feel free to contribute to Maco](https://gitlab.com/pop-planet/maco).

hargoniX, 2019-06-12:
It's been mentioned above that
> # END OF OFFICIAL VOTES
> Only items listed above this line will be considered for the official vote.

And also that
> Only selections proposed by the Embedded WG will be considered.

So discussing your favourite chat service here won't really yield any useful results.

IGBC, 2019-06-13:
If I know anything about which number is bigger than the other, then the rankings are:

1. Matrix
2. IRC @ Frenode
3. Discord

The 3 vote rule has not come into effect. 

3.

caemor, 2019-06-13:
The vote is still ongoing for another 10 days ;-)

jamesmunns, 2019-06-23:
Votes as of 00:30 CEST:

* Discord: 74
* Gitter: 11
* IRC (Freenode): 103
* Matrix: 160
* Zulip: 52

It looks like Matrix has gotten the top vote by more than three votes.

We'll be implementing these changes over the next few weeks, and will discuss how to implement this in the upcoming meeting on IRC on Tuesday at 20:00 CEST. See #362 for the agenda. Stay tuned for a switch-over date from IRC to Matrix in the future.

Thanks all for voting!

eddyp, 2019-06-24:
> 
> 
> Votes as of 00:30 CEST:
> 
>     * Discord: 74
> 
>     * Gitter: 11
> 
>     * IRC (Freenode): 103
> 
>     * Matrix: 160
> 
>     * Zulip: 52
> 
> 
> It looks like Matrix has gotten the top vote by more than three votes.

It seems people were under the impression that downvotes 👎  (or upvotes 👍 - downvotes 👎 ) also accounted for something.

I wonder how many people would have felt cheated if Discord had the most upvotes and most down votes, but Matrix had less upvotes than Discord, but way less down votes.

Since people seem to have felt downvotes acount for something, maybe the rules should be changed to have a classification accounting for downvotes somehow?

therealprof, 2019-06-24:
@eddyp Huh? The rules were pretty clear. Changing that after the fact would be cheating to the people who are able to read and acted accordingly.

andre-richter, 2019-06-24:
@eddyp I really don't see how the initial post in this thread could have given this impression. Its formulation is not ambiguous in any way.

f0x52, 2019-06-24:
@eddyp Even when counting upvotes :+1: - downvotes :-1:, Matrix is still the clear winner, with 142 compared to the next follow-up, IRC at 58 (calculated with current results, since downvotes weren't saved at the end of the voting period)

closed 2019-06-24 by jamesmunns

maxidorius, 2019-06-28:
A bit late, but here is a research document on privacy regarding self-hosted Matrix installs when relying more or less on default settings: https://gist.github.com/maxidorius/5736fd09c9194b7a6dc03b6b8d7220d0

MatMaul, 2019-07-01:
And here is a recent update on that from the Matrix team:
https://matrix.org/blog/2019/06/30/tightening-up-privacy-in-matrix

ara4n, 2019-11-19:
...and for the sake of completeness, https://matrix.org/blog/2019/09/27/privacy-improvements-in-synapse-1-4-and-riot-1-4/ is where Matrix finished shipping the refinements to the default configs which were called into question.