rust-embedded/wg #359: Embedded WG Meeting, Jun 18th

jamesmunns, 2019-06-17:
[8-9 PM CET][0] (Berlin time) on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2019-06-18,420,b8jj

[Public Google calendar that includes these weekly meetings](https://calendar.google.com/calendar?cid=NzhmdGJsMGQ1cjA2YzM5dDJ1MDU0ODBsbm9AZ3JvdXAuY2FsZW5kYXIuZ29vZ2xlLmNvbQ)

Agenda.

- Reminders / announcements
- [Issues nominated for discussion, if any](https://github.com/rust-embedded/wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

jamesmunns, 2019-06-17:
NB @therealprof, I will likely not be attending this meeting

therealprof, 2019-06-18:
This meeting has happened! 🎉

commit c95b9134 (2019-06-19): Added meeting notes for 2019-06-18

Closes #359
commit 2965fab9 (2019-06-19): Merge #360

360: Added meeting notes for 2019-06-18 r=Disasm a=therealprof

Closes #359
closed 2019-06-19 by bors[bot]